adjacentlink/emane #19: virtual transport unknown protocol destination mapping always broadcast

sgalgano, 2015-05-03:
Reported via email by Adrian Granados:

I think I found the problem and it has to do with how EMANE handles frames with “unknown protocol” types. 

If you see the method “parseFrame” in src/transports/common/ethernettransport.cc, EMANE is assuming that the destination NEM for a frame that contains an “unknown protocol” type is NEM_BROADCAST_MAC_ADDRESS. I don’t believe EMANE should make such assumption. 

In my particular scenario, the proprietary protocol I’m testing has an ether type 0x2800 (unknown for EMANE), but the source and destination MAC addresses are interpreted as normal and can be unicast or broadcast. Because EMANE assumes broadcast, it processes all frames with unicast and broadcast MAC destinations using the multicast rate. In my scenario,  multicastrate=2 and unicastrate=8, which results in nodes A and B seeing each other using broadcast but not using unicast frames. This is why “ping” works (nodes CANNOT ping each other) but the “unknown protocol” doesn’t (nodes CAN communicate with each other).

My suggestion is to change:

```C++
   // unknown protocol
     default:
       LOGGER_VERBOSE_LOGGING(pPlatformService_->logService(),
                              DEBUG_LEVEL,
                              "TRANSPORTI %03d EthernetTransport::%s allow unknown protocol %02X",
                              id_,
                              __func__,
                              u16ethProtocol);

       // use broadcast mac addr
       rNemDestination = NEM_BROADCAST_MAC_ADDRESS;

       // set dscp to 0
       rDspc = 0;
```

to:
```C++
    // unknown protocol
    default:
      	LOGGER_VERBOSE_LOGGING(pPlatformService_->logService(),
                              DEBUG_LEVEL,
                              "TRANSPORTI %03d EthernetTransport::%s allow unknown protocol %02X",
                              id_,
                              __func__,
                              u16ethProtocol);

       // broadcast always mode
        if(bBroadcastMode_)
           {
             rNemDestination = NEM_BROADCAST_MAC_ADDRESS;
           }
        // use ether dst
        else
          {
            rNemDestination = Utils::ethaddr4_to_id(&pEthHeader->dst);
          }

        // set dscp to 0
        rDspc = 0;
```

After making the change above, the emulation works as expected.

sgalgano, 2015-05-03:
The proposed patch replaces the always use NEM_BROADCAST_MAC_ADDRESS for unknown protocol logic with the same look-up order as the known protocol cases.

```diff
diff --git a/src/transports/common/ethernettransport.cc b/src/transports/common/ethernettransport.cc
index 4f5fe02..6c96e62 100644
--- a/src/transports/common/ethernettransport.cc
+++ b/src/transports/common/ethernettransport.cc
@@ -267,9 +267,22 @@ int EMANE::Transports::Ethernet::EthernetTransport::parseFrame(const Utils::Ethe
                               __func__, 
                               u16ethProtocol);
 
-       // use broadcast mac addr
-       rNemDestination = NEM_BROADCAST_MAC_ADDRESS;
-
+       // broadcast always mode
+       if(bBroadcastMode_) 
+         {
+           rNemDestination = NEM_BROADCAST_MAC_ADDRESS;
+         }
+       // check arp cache
+       else if (bArpCacheMode_)
+         {
+           rNemDestination = lookupArpCache(&pEthHeader->dst);
+         }
+       // use ether dst
+       else 
+         {
+           rNemDestination = Utils::ethaddr4_to_id(&pEthHeader->dst);
+         }
+       
        // set dscp to 0
        rDspc = 0;
 
```

jgiovatto, 2015-05-04:
Since the ether type is not any of the above (IPv4/IPv6) then I think it makes more sense to just look at the last 2 bytes for the dst NEM id as EMANE generally does. This should cover the P2P and Broadcast cases and not conflict with IP Multicast. 
suggest changing:
rNemDestination = Utils::ethaddr4_to_id(&pEthHeader->dst);
to
rNemDestination = ACE_NTOHS(pEthHeader->words.word3);

sgalgano, 2015-05-04:
That makes sense. Here is the updated proposed patch:

```diff
diff --git a/src/transports/common/ethernettransport.cc b/src/transports/common/ethernettransport.cc
index 4f5fe02..0a8c921 100644
--- a/src/transports/common/ethernettransport.cc
+++ b/src/transports/common/ethernettransport.cc
@@ -267,8 +267,21 @@ int EMANE::Transports::Ethernet::EthernetTransport::parseFrame(const Utils::Ethe
                               __func__, 
                               u16ethProtocol);
 
-       // use broadcast mac addr
-       rNemDestination = NEM_BROADCAST_MAC_ADDRESS;
+       // broadcast always mode
+       if(bBroadcastMode_)
+         {
+           rNemDestination = NEM_BROADCAST_MAC_ADDRESS;
+         }
+       // check arp cache
+       else if (bArpCacheMode_)
+         {
+           rNemDestination = lookupArpCache(&pEthHeader->dst);
+         }
+       // use the last 2 bytes of the ethernet destination
+       else
+         {
+           rNemDestination = ACE_NTOHS(pEthHeader->dst.words.word3);
+         }
 
        // set dscp to 0
        rDspc = 0;
```

sgalgano, 2015-05-04:
This patch works with the current virtual transport default configuration provided you have other traffic that would cause ARP requests/responses. The default virtual transport configuration has *arpcacheenable* set to *on*.  This allows out-of-the-box interoperability between the virtual and raw transports. 

The arp cache functionality will set the destination nem id to NEM_BROADCAST_MAC_ADDRESS when it cannot find the destination ethernet mac address in the cache. If you are not generating IP traffic, the arp cache will remain empty. In this situation, when using the virtual transport, set *arpcacheenable* to *off*, which will cause the last two bytes of the destination ethernet mac address to be used as the destination nem for point-to-point traffic.

```XML
<!DOCTYPE transport SYSTEM "file:///usr/share/emane/dtd/transport.dtd">
<transport library="transvirtual">
  <param name="arpcacheenable" value="off" />
</transport>
```

This will only work when using virtual transports. The raw transport requires at least one associated arp request/response per destination ethernet mac address.

closed 2015-05-04 by sgalgano

adriangm20, 2015-05-05:
Just to confirm the patch works in the original scenario where the bug was found.

sgalgano, 2015-05-05:
Thanks for testing.