adjacentlink/emane #69: Compile errors on Ubuntu 16.04.1 i686

tomgoff, 2016-09-13:
Errors like the following occur when building the develop branch on Ubuntu 16.04.1 i686 using g++ version 5.4.0:
```
eventstatisticpublisher.cc:115:49: error: call of overloaded 'Any(<brace-enclosed initializer list>)' is ambiguous
                                           Any{0L},
                                                 ^
In file included from ../../include/emane/statistic.h:37:0,
                 from ../../include/emane/statisticnumeric.h:37,
                 from eventstatisticpublisher.h:37,
                 from eventstatisticpublisher.cc:33:
../../include/emane/any.h:142:5: note: candidate: EMANE::Any::Any(const EMANE::Any&)
     Any(const Any & rhs);
     ^
../../include/emane/any.h:137:14: note: candidate: EMANE::Any::Any(const bool&)
     explicit Any(const bool & bValue);
              ^
../../include/emane/any.h:132:14: note: candidate: EMANE::Any::Any(const string&)
     explicit Any(const std::string & sValue);
              ^
../../include/emane/any.h:127:14: note: candidate: EMANE::Any::Any(const char*)
     explicit Any(const char * pzValue);
              ^
../../include/emane/any.h:117:14: note: candidate: EMANE::Any::Any(double)
     explicit Any(double dValue);
              ^
../../include/emane/any.h:112:14: note: candidate: EMANE::Any::Any(float)
     explicit Any(float fValue);
              ^
../../include/emane/any.h:107:14: note: candidate: EMANE::Any::Any(uint8_t)
     explicit Any(std::uint8_t u8Value);
              ^
../../include/emane/any.h:102:14: note: candidate: EMANE::Any::Any(int8_t)
     explicit Any(std::int8_t i8Value);
              ^
../../include/emane/any.h:97:14: note: candidate: EMANE::Any::Any(uint16_t)
     explicit Any(std::uint16_t u16Value);
              ^
../../include/emane/any.h:92:14: note: candidate: EMANE::Any::Any(int16_t)
     explicit Any(std::int16_t i16Value);
              ^
../../include/emane/any.h:87:14: note: candidate: EMANE::Any::Any(uint32_t)
     explicit Any(std::uint32_t u32Value);
              ^
../../include/emane/any.h:82:14: note: candidate: EMANE::Any::Any(int32_t)
     explicit Any(std::int32_t i32Value);
              ^
../../include/emane/any.h:77:14: note: candidate: EMANE::Any::Any(uint64_t)
     explicit Any(std::uint64_t u64Value);
              ^
../../include/emane/any.h:72:14: note: candidate: EMANE::Any::Any(int64_t)
     explicit Any(std::int64_t i64Value);
              ^
```

The attached patch is a possible fix.
[emane-any-ambiguous.patch.txt](https://github.com/adjacentlink/emane/files/470630/emane-any-ambiguous.patch.txt)

sgalgano, 2016-12-14:
There are other issues related to 32 bit support concerning log statements which include 64 bit integers. We have not built emane for 32 bit systems for a number of years now.

tomgoff, 2017-04-19:
If 32-bit systems are not supported, would it make sense to have configure-time warning or error?

Possibly something like this might work:
```
AC_CHECK_SIZEOF([int])
AC_CHECK_SIZEOF([long])
if test $ac_cv_sizeof_int -eq $ac_cv_sizeof_long
then
AC_MSG_WARN([unsupported system detected (non-64-bit?)])
fi
```