adjacentlink/emane #78: Terminal Issue after emane demonstration

mdcovarr, 2017-11-21:
Hello, I am a new user to EMANE, I recently got the framework to run on my virtual machine of Ubuntu 16.04 LTS. Ran into a few issues but managed to get the software running and went through the demonstrations. The problem is that after I start a demonstration:
`$ sudo ./demo-start` 
and then try to open another terminal window I am presented with the error correlating to the following image where I am not able to open another window. In order to be able to open I need to restart the virtual machine again to be able to open a terminal again. Does anyone have an idea as to why this occurs? Thanks!

[image]

sgalgano, 2017-11-29:
Something changed after lxc version 2.0.5 causing the issue you are experiencing. The below  [emane-tutorial](https://github.com/adjacentlink/emane-tutorial)  `develop` branch patch makes the appropriate container config modification and adds an `lxc.hook.autodev`.

```patch
diff --git a/scripts/democtl-host b/scripts/democtl-host
index 92bbfe2..1b09f3e 100755
--- a/scripts/democtl-host
+++ b/scripts/democtl-host
@@ -358,10 +358,21 @@ lxc.tty = 1
 lxc.pts = 128
 lxc.cgroup.devices.allow = a
 
-lxc.autodev = 0
+lxc.autodev = 1
+lxc.hook.autodev = $topdir/$demoid/persist/$nodeid/var/run/lxc.hook.autodev.sh
 
 EOF
 
+            cat <<EOF > persist/$nodeid/var/run/lxc.hook.autodev.sh
+#!/bin/bash -
+if ! [ -c /dev/net/tun ]; then
+ mkdir -p /dev/net
+ mknod -m 666 /dev/net/tun c 10 200
+fi
+
+EOF
+            chmod a+x persist/$nodeid/var/run/lxc.hook.autodev.sh
+
             if [[ $lxcmajorversion -ne 0 ]]
             then
                 echo "lxc.aa_profile = unconfined" >> persist/$nodeid/var/run/lxc.conf.$nodeid

```
In the future, issues related to the [emane-tutorial](https://github.com/adjacentlink/emane-tutorial) should be created using that project's [issues](https://github.com/adjacentlink/emane-tutorial/issues). I'll open up a corresponding tutorial issue to merge this patch in. Some tweaks may be necessary to support distributions with lxc versions 2.0.5 and older.

mdcovarr, 2017-12-01:
Thank you, this did the trick for me. I apologize for not creating this issue under the emane-tutorial project and will definitely so so in future issues.

closed 2017-12-04 by sgalgano