adjacentlink/emane #79: Downstream and upstream packet length is limited to 16 bits.

tomgoff, 2018-01-16:
Commit b88fa0a added support for sending and receiving OTA messages larger than 65535 bytes, but the length processing of downstream and upstream packets did not change.

The attached patch allows sending and receiving downstream and upstream packets that are larger than 65535 bytes.
[emane-downstream-upstream-packet-length.patch.txt](https://github.com/adjacentlink/emane/files/1636125/emane-downstream-upstream-packet-length.patch.txt)

sgalgano, 2018-01-16:
This patch allows you to prepend and remove *length prefix frames* for individual packet headers that are greater than 65535. Packets and headers are capable of being larger.  We made a conscious decision not to increase the maximum supported length prefix frame size so not to change (break) the packet API. You can still add larger headers with or without your own larger length prefix frame.

You may have to modify how you are creating your packets if this is a problem.

tomgoff, 2018-01-23:
Fair enough regarding backward compatibility.

It might be worth considering adding helper methods like:
```
void DownstreamPacket::prependLengthPrefixFramingLong(std::uint32_t u32Length);

std::uint32_t UpstreamPacket::stripLengthPrefixFramingLong();
```
so models that send/receive large OTA messages don't have to implement their own functionality and can have similar logic to the standard EMANE models.

sgalgano, 2018-01-26:
Thanks for the suggestion.

closed 2018-01-26 by sgalgano