adjacentlink/emane #85: Euler Angles Order & Antenna-Platform Offsets

HaasCM, 2018-04-16:
Hi, I've got two questions regarding the order of the Euler angles to represent free body orientation and the NEU coordinate system to describe the location of the Antenna's aperture in the platform's coordinate system. 

Is the order in which Euler angles are applied to the platform's orientation pitch, roll, then yaw?

and,

Is the placement of the antenna's aperture NEU location within the platform's coordinate system in meters? Does this location assume that the platform velocity vector has an azimuth and elevation of zero?

patelkb, 2018-04-17:
1.  The rotations are applied in the order of roll/pitch/yaw to
represent the platform's local frame of reference.

2.  The platform's velocity vector is accounted for by adjusting the Yaw
and Pitch angels prior to doing the rotations.

3.  The antenna placement parameters are in meters.

kb

Kaushik B. Patel
Adjacent Link LLC

On 04/16/2018 06:53 PM, Cody Haas wrote:
> Hi, I've got two questions regarding the order of the Euler angles to represent free body orientation and the NEU coordinate system to describe the location of the Antenna's aperture in the [...]

HaasCM, 2018-04-17:
Thanks for the response, that really clears things up! One last question, for the placement of the antenna, is the local ENU system assuming that the platform has a heading of 0? And is the origin of the ENU system the center of rotation of the platform?

patelkb, 2018-04-17:
The velocity vector defines the platform's direction relative to True
North so it doesn't imply 0 heading.

The NEU reference frame is based on the node's location and Euler angle
rotations are performed along the platform's vertical (down), transverse
and longitudinal axes.

Kaushik B. Patel
Adjacent Link LLC

On 04/17/2018 03:16 PM, Cody Haas wrote:
> Thanks for the response, that really clears things up! One last question, for the placement of the antenna, is the local ENU system assuming that the platform has a heading of 0? And is the origin of the ENU system the center of rotation of the platform?
>

HaasCM, 2018-04-27:
That clears up a lot and helps me wrap my head around these concepts. Thanks for all of the work you all put in on EMANE!