adjacentlink/emane #87: Dynamic NEM creation

rjmccabe3701, 2018-06-19:
We are transitioning an NRL CORE-based environment -- where each node has its own local ``emane`` process -- to our own docker-based environment.  In this environment, we were hoping to allow for the single-process EMANE approach showcased in [demo 2](https://github.com/adjacentlink/emane-tutorial/wiki/Demonstration-2).  We used such a configuration a few years ago -- it was nice because it allowed for higher throughput link calculations (could use a GPU to better "batch" computation).  

The only issue I see with this approach is (at least it seems) its not possible to dynamically create or destroy NEMs after the EMANE process initializes.  Does this functionality exist?  If not, would it be difficult to add?

sgalgano, 2018-06-20:
I'm not sure I totally follow your question. It is possible to build your own application that links with `libemane`. Your application can create NEMs based on any configuration mechanism. That is pretty much what the `emane` application does.

I'd start by looking at [NEMBuilder](https://github.com/adjacentlink/emane/blob/master/include/emane/application/nembuilder.h) and [NEMManager](https://github.com/adjacentlink/emane/blob/master/include/emane/application/nemmanager.h) to see how NEMs are instantiated and managed. Then take a look at [Emulator](https://github.com/adjacentlink/emane/blob/master/src/libemane/emulator.h) and [Main](https://github.com/adjacentlink/emane/blob/master/src/libemane/main.cc) for the startup flow of the emulator.

rjmccabe3701, 2018-06-21:
My question was more concerning does EMANE natively (without writing custom C++ code) dynamic NEM creation/deletion?  I was hoping that a protocol buffer-based API could be used to communicate with a running EMANE process.  For instance, it would be neat if there were a utility in the [scripts](https://github.com/adjacentlink/emane/tree/master/src/python/scripts) directory called ``emaneevent-nemcontrol`` or something.  And this could communicate with the NemDirector in the running emane process, and direct it to create/destroy nems.  Of course the protocol-buffer based protocol to create a new NEM would require all the information (mac/phy plugin names, configuration, etc) that is provided by the xml files (used when running ``emane`` via the normal command line method). 

The reason I ask: I believe I heard from a co-worker that such a feature exists in the newer EMANE releases (I haven't used emane since v0.9.3 -- I'm just getting ramped back up on its capabilities), was just asking the experts before I do a deep-dive in the source.

I agree that adding this functionality to this repository would be a lot of work and could be accomplished by just using the tools provided by ``libemane``.  I will look into your suggestions and write some custom code to do what I want.  Thanks!

closed 2018-09-04 by sgalgano