adjacentlink/emane #89: Earth blockage?

cSchubes, 2018-08-31:
When propagationmode is set to freespace, does EMANE take into account the curvature of the earth?

For example, if I place two NEMs quite far from each other in terms of latitude and longitude, but give them very powerful radios, will the curvature of the earth prevent them from linking? Or is the simulation purely 2-D?

Thanks!

sgalgano, 2018-08-31:
The physical layer will drop packets that are below the horizon, regardless of the selected propagation model provided both transmitter and receiver positions are known.

cSchubes, 2018-08-31:
I have two nodes, both with known locations and on opposite sides of the globe (both at latitude of 40, with one at long of -74 and the other at long of 74) and a radio power of 80 dBm. A link is still somehow being established despite this. My setup is almost exactly as Demonstration 1 is, with a few minor changes to have 2 nodes instead of 10, and to make the nodes more powerful and use freespace propagation. 

What could the reason for this be?

cSchubes, 2018-08-31:
output of `emanesh node-1 get table nems phy`
```
nem 1   phy AntennaProfileEventInfoTable
| NEM | Antenna Profile | Antenna Azimuth | Antenna Elevation |

nem 1   phy BroadcastPacketAcceptTable0
| NEM | Num Pkts Tx | Num Bytes Tx | Num Pkts Rx | Num Bytes Rx |
| 1   | 85          | 13630        | 0           | 0            |
| 2   | 0           | 0            | 33          | 9102         |

nem 1   phy BroadcastPacketDropTable0
| NEM | Out-of-Band | Rx Sensitivity | Propagation Model | Gain Location | Gain Horizon | Gain Profile | Not FOI | Spectrum Clamp | Fade Location | Fade Algorithm | Fade Select |
| 2   | 0           | 0              | 50                | 0             | 0            | 0            | 0       | 0              | 0             | 0              | 0           |

nem 1   phy EventReceptionTable
| Event | Total Rx |
| 100   | 1        |

nem 1   phy FadingSelectionInfoTable
| NEM | Model |

nem 1   phy LocationEventInfoTable
| NEM | Latitude | Longitude  | Altitude | Pitch | Roll | Yaw | Azimuth | Elevation | Magnitude |
| 1   | 40.0     | -74.523518 | 3.0      | 0.0   | 0.0  | 0.0 | 0.0     | 0.0       | 0.0       |
| 2   | 40.0     | 74.0       | 3.0      | 0.0   | 0.0  | 0.0 | 0.0     | 0.0       | 0.0       |

nem 1   phy PathlossEventInfoTable
| NEM | Forward Pathloss | Reverse Pathloss |

nem 1   phy ReceivePowerTable
| NEM | Frequency  | Rx Power       | Last Packet Time |
| 2   | 2347000000 | -99.3377475998 | 1535748308.68    |

nem 1   phy UnicastPacketAcceptTable0
| NEM | Num Pkts Tx | Num Bytes Tx | Num Pkts Rx | Num Bytes Rx |

nem 1   phy UnicastPacketDropTable0
| NEM | Out-of-Band | Rx Sensitivity | Propagation Model | Gain Location | Gain Horizon | Gain Profile | Not FOI | Spectrum Clamp | Fade Location | Fade Algorithm | Fade Select |
```
output of `emanesh node-2 get table nems phy`
```
nem 2   phy AntennaProfileEventInfoTable
| NEM | Antenna Profile | Antenna Azimuth | Antenna Elevation |

nem 2   phy BroadcastPacketAcceptTable0
| NEM | Num Pkts Tx | Num Bytes Tx | Num Pkts Rx | Num Bytes Rx |
| 1   | 0           | 0            | 257         | 83766        |
| 2   | 305         | 87011        | 0           | 0            |

nem 2   phy BroadcastPacketDropTable0
| NEM | Out-of-Band | Rx Sensitivity | Propagation Model | Gain Location | Gain Horizon | Gain Profile | Not FOI | Spectrum Clamp | Fade Location | Fade Algorithm | Fade Select |
| 1   | 0           | 0              | 48                | 0             | 0            | 0            | 0       | 0              | 0             | 0              | 0           |

nem 2   phy EventReceptionTable
| Event | Total Rx |
| 100   | 1        |

nem 2   phy FadingSelectionInfoTable
| NEM | Model |

nem 2   phy LocationEventInfoTable
| NEM | Latitude | Longitude  | Altitude | Pitch | Roll | Yaw | Azimuth | Elevation | Magnitude |
| 1   | 40.0     | -74.523518 | 3.0      | 0.0   | 0.0  | 0.0 | 0.0     | 0.0       | 0.0       |
| 2   | 40.0     | 74.0       | 3.0      | 0.0   | 0.0  | 0.0 | 0.0     | 0.0       | 0.0       |

nem 2   phy PathlossEventInfoTable
| NEM | Forward Pathloss | Reverse Pathloss |

nem 2   phy ReceivePowerTable
| NEM | Frequency  | Rx Power       | Last Packet Time |
| 1   | 2347000000 | -99.3377475998 | 1535748413.08    |

nem 2   phy UnicastPacketAcceptTable0
| NEM | Num Pkts Tx | Num Bytes Tx | Num Pkts Rx | Num Bytes Rx |

nem 2   phy UnicastPacketDropTable0
| NEM | Out-of-Band | Rx Sensitivity | Propagation Model | Gain Location | Gain Horizon | Gain Profile | Not FOI | Spectrum Clamp | Fade Location | Fade Algorithm | Fade Select |
```
output of `emanesh node-1 get config nems phy`
```
nem 1   phy  bandwidth = 1000000
nem 1   phy  fading.model = none
nem 1   phy  fading.nakagami.distance0 = 100.0
nem 1   phy  fading.nakagami.distance1 = 250.0
nem 1   phy  fading.nakagami.m0 = 0.75
nem 1   phy  fading.nakagami.m1 = 1.0
nem 1   phy  fading.nakagami.m2 = 200.0
nem 1   phy  fixedantennagain = 0.0
nem 1   phy  fixedantennagainenable = True
nem 1   phy  frequency = 2347000000
nem 1   phy  frequencyofinterest = 2347000000
nem 1   phy  noisebinsize = 20
nem 1   phy  noisemaxclampenable = False
nem 1   phy  noisemaxmessagepropagation = 200000
nem 1   phy  noisemaxsegmentduration = 1000000
nem 1   phy  noisemaxsegmentoffset = 300000
nem 1   phy  noisemode = outofband
nem 1   phy  propagationmodel = freespace
nem 1   phy  subid = 2
nem 1   phy  systemnoisefigure = 4.0
nem 1   phy  timesyncthreshold = 10000
nem 1   phy  txpower = 80.0
```
output of `emanesh node-2 get config nems phy`
```
nem 2   phy  bandwidth = 1000000
nem 2   phy  fading.model = none
nem 2   phy  fading.nakagami.distance0 = 100.0
nem 2   phy  fading.nakagami.distance1 = 250.0
nem 2   phy  fading.nakagami.m0 = 0.75
nem 2   phy  fading.nakagami.m1 = 1.0
nem 2   phy  fading.nakagami.m2 = 200.0
nem 2   phy  fixedantennagain = 0.0
nem 2   phy  fixedantennagainenable = True
nem 2   phy  frequency = 2347000000
nem 2   phy  frequencyofinterest = 2347000000
nem 2   phy  noisebinsize = 20
nem 2   phy  noisemaxclampenable = False
nem 2   phy  noisemaxmessagepropagation = 200000
nem 2   phy  noisemaxsegmentduration = 1000000
nem 2   phy  noisemaxsegmentoffset = 300000
nem 2   phy  noisemode = outofband
nem 2   phy  propagationmodel = freespace
nem 2   phy  subid = 2
nem 2   phy  systemnoisefigure = 4.0
nem 2   phy  timesyncthreshold = 10000
nem 2   phy  txpower = 80.0
```

sgalgano, 2018-09-01:
You discovered a bug in the [GainManager](https://github.com/adjacentlink/emane/blob/v1.2.2/src/libemane/gainmanager.cc#L143). Thanks for reporting it.

The variable holding the distance between nodes, which is used in the below horizon check, is only being populated when antenna profiles are in use. The fix is straightforward:

```diff
diff --git a/src/libemane/gainmanager.cc b/src/libemane/gainmanager.cc
index 4e92587..1551304 100644
--- a/src/libemane/gainmanager.cc
+++ b/src/libemane/gainmanager.cc
@@ -92,7 +92,6 @@ EMANE::GainManager::determineGain(NEMId transmitterId,
   double dRxAntennaBlockagedBi{};
   double dTxAntennaBlockagedBi{};
 
-  double dDistanceMeters{};
   PositionNEU antennaPlacement{};
 
   LOGGER_VERBOSE_LOGGING_FN_VARGS(*LogServiceSingleton::instance(),
@@ -140,8 +139,6 @@ EMANE::GainManager::determineGain(NEMId transmitterId,
                                       locationPairInfo.getLocalPOV(),
                                       localAntennaPlacement_);
                 
-          dDistanceMeters = std::get<2>(direction);
-                
           // adjust the direction azimuth and elvation based on the antenna pointing azimuth and elvation
           auto lookupAngles = 
             Utils::calculateLookupAngles(std::get<0>(direction),
@@ -271,6 +268,7 @@ EMANE::GainManager::determineGain(NEMId transmitterId,
 
   const auto & localPosition = locationPairInfo.getLocalPOV().getPosition();
   const auto & remotePosition = locationPairInfo.getRemotePOV().getPosition();
+  double dDistanceMeters{locationPairInfo.getDistanceMeters()};
   
   // check if antennas are below the horizon
   if(!locationPairInfo == false && 
```

We need to review the results of the gain scenario test cases with this change. Then we'll push it upstream.

sgalgano, 2018-09-01:
This fix has been pushed to `develop` and will be in the next release. Thanks.

sgalgano, 2018-09-04:
My suggestion would be to build the `develop` branch.

closed 2018-09-04 by sgalgano

cSchubes, 2018-09-05:
Hey again -

When I try to build the develop branch it is failing with
```
In file included from nembuilder.cc:35:0:
../../include/emane/application/nembuilder.h:255:44: fatal error: emane/application/nembuilder.inl: No such file or directory
compilation terminated
```
I know that develop is not a stable branch and thus errors can be expected. But I really need to have access to the fix this issue describes! I'm following [these](https://github.com/adjacentlink/emane/wiki/Build#ubuntu-14041604-lts) instructions for building on Ubuntu 16.04 LTS.

Sorry to keep bugging you guys with this stuff - thanks again for everything so far.