adjacentlink/emane #90: Changing the frequency of a NEM during emulation

cSchubes, 2018-09-04:
Hi,

My use case requires the ability to update the frequency a NEM communicates on throughout emulation of the network. I see that on the PHY layer, frequency is marked as `Parameter not running state modifiable`, which makes sense. 

The [EMANE User Manual](https://downloads.pf.itd.nrl.navy.mil/docs/emane/emane.pdfl) from 0.8.1 states that "The Tx Control API provides the MAC Layer with the ability to override default PHY Layer configuration for transmit power, message duration, transmit frequency and antenna pointing as required." (Page 69) This Control API is mentioned throughout the manual, but I don't see anything about how to use it. I've checked all occurrences in the document of 'Control API' to no avail. Maybe I am missing something?

The closest I've found is the _transmissioncontrolmap_ parameter for the RF Pipe MAC Layer, which says is "Defines the data rate, frequency, and power level to be used by the PHY Layer for all transmissions to a specified node." (Page 88). However, this appears to be a deprecated parameter - it is not present in the newest release.

I realize that the User Manual is from 0.8.1 and is not up to date, and have done a fairly exhaustive search of the 'newer documentation' linked on the [EMANE Home Page](https://www.nrl.navy.mil/itd/ncs/products/emane) (which is the GitHub hosted Wiki) but can't seem to find anything about the Control API or the _transmissioncontrolmap_ parameter.

Has this feature been deprecated? Is frequency of a NEM completely static for the runtime of the emulation now?

Any help is greatly appreciated!!

sgalgano, 2018-09-04:
That parameter has been removed for quite some time. I would suggest not referencing any documentation < 1.0.1. 

The only open source model that supports dynamic frequency change out of the box without changing code, is the [TDMA model](https://github.com/adjacentlink/emane/wiki/TDMA-Model).

cSchubes, 2018-09-04:
I had initially went with RF Pipe, since I will only have one user per frequency. I imagine I will be able to use TDMA for the same effect, but with the ability to change frequencies over time. Is that correct? Is there any reason using the TDMA model to simulate a radio which does not take advantage of TDMA features would be ill advised? I would like to avoid writing my own model if possible.

cSchubes, 2018-09-04:
Also, the key feature I need is the ability to change the frequency during runtime WITHOUT knowing ahead of time what those changes will be. I cannot predefine them in my case.

sgalgano, 2018-09-04:
If the radio channel access scheme is not important to your experiment, it should not matter. RF Pipe does not do channel access.

You can use existing `emane.events` python module code to dynamically build and send TDMA schedules. You will have to coordinate schedules between transmitters and receivers since you need to set the receive side slot frequency as well.

cSchubes, 2018-09-04:
Gotcha. I should be able to use a single frame, single slot schedule to emulate a simple, non-TDMA radio and update the schedule over time. 

Thank you for all the help!!

closed 2018-12-21 by sgalgano