adjacentlink/emane #91: develop branch make deb error: missing emane/application/nembuilder.inl

sgalgano, 2018-09-05:
Reported by @cSchubes.

When I try to build the develop branch it is failing with
```
In file included from nembuilder.cc:35:0:
../../include/emane/application/nembuilder.h:255:44: fatal error: emane/application/nembuilder.inl: No such file or directory
compilation terminated
```

cSchubes, 2018-09-05:
Thanks - wasn't sure if I should report this as a new issue, since its to do with the develop branch.

sgalgano, 2018-09-05:
We use gitflow for branch management. The `develop` branch should always be stable.

`nembuilder.inl ` is missing from `emaneinc_HEADERS` preventing `make deb` and `make rpm` from working. Non package build works. The following patch remedies the issue.

```diff
diff --git a/include/emane/application/Makefile.am b/include/emane/application/Makefile.am
index 720b3ce..3a224fb 100644
--- a/include/emane/application/Makefile.am
+++ b/include/emane/application/Makefile.am
@@ -9,6 +9,7 @@ emaneinc_HEADERS =                       \
  libemane.h                              \
  logger.h                                \
  nembuilder.h                            \
+ nembuilder.inl                          \
  nem.h                                   \
  nemmanager.h                            \
  statisticcontroller.h                   \
```

Patch tested on Fedora 27 and Ubuntu 16.04.

sgalgano, 2018-09-05:
Thanks for reporting the issue.

closed 2018-09-05 by sgalgano