adjacentlink/emane #95: dynamic mcs index

ed-silvus, 2019-03-28:
I would like to have the mcs index change dynamically according to the sinr. Is there a suggested way to modify the ieee80211 model in order to achieve this?

sgalgano, 2019-03-29:
The complexity of the modification depends on your exact experiment goals. If you want to implement a feature with per destination MCS indexes, it will be more complex than if you just change it for all unicast and broadcast traffic. Both would require closed loop SINR info from receiver to transmitter to make changes at the transmitter.

ed-silvus, 2019-03-29:
I would like to change the unicast rate per destination. Is the getMessageDurationMicroseconds function the correct area of the code to look at?

sgalgano, 2019-04-01:
Yes

closed 2019-06-07 by sgalgano