adjacentlink/emane #96: Ubuntu 16.04- building distribution packages from EMANE source

enoorani, 2019-04-05:
### I have difficulty installing EMANE and I'm not entirely sure where the problem is. I would truly appreciate any help.

Ubuntu 16.04-
 building distribution packages from EMANE source:
Step 3: the command "make deb WITHOUT_PYTHON3=1" throws the following error:

make[2]: Leaving directory '/home/seiluser/emane/.debbuild/emane-1.2.4'
   dh_shlibdeps -O--parallel
dpkg-shlibdeps: error: no dependency information found for /usr/local/lib/libprotobuf.so.9 (used by debian/emane-libs/usr/lib/libemane.so)
Hint: check if the library actually comes from a package.
dh_shlibdeps: dpkg-shlibdeps -Tdebian/emane-libs.substvars debian/emane-libs/usr/lib/libemanexmlparser.so debian/emane-libs/usr/lib/libemane.so returned exit code 2
debian/rules:49: recipe for target 'binary' failed
make[1]: *** [binary] Error 2
make[1]: Leaving directory '/home/seiluser/emane/.debbuild/emane-1.2.4'
dpkg-buildpackage: error: debian/rules binary gave error exit status 2
Makefile:892: recipe for target 'deb' failed
make: *** [deb] Error 2

[image]

also I tried to install  EMANE from pre-built distribution packages but
in Step2, the command "dpkg -i *.deb" throws this error:

dpkg: dependency problems prevent configuration of python3-emane:
 python3-emane depends on python3-protobuf; however:
  Package python3-protobuf is not installed.

[image]

I did  pip list | grep python3- and it returns 
python3-protobuf (2.5.0)

[image]

eschreiber-alink, 2019-04-05:
> Screenshot from 2019-04-04 20-20-50
> image

Have a look at the build directions for Ubuntu 16.04 on the Wiki,
especially the first step for installing dependencies from the Ubuntu repos:

https://github.com/adjacentlink/emane/wiki/Build#ubuntu-14041604-lts

enoorani, 2019-04-05:
@eschreiber-alink : Thanks for your comment, but all the dependencies are already installed! Am I missing something here?
[image]

eschreiber-alink, 2019-04-05:
I recommend removing PIP installed dependencies. EMANE compiles cleanly
on Ubuntu 16.04 with the wiki directions as long as you haven't also
redundantly installed items from PIP over the Ubuntu repo versions.

"/usr/local/lib" in your error message is a red flag.

enoorani, 2019-04-05:
@eschreiber-alink uninstalled the pip installed protobuf package, but the error is still there. any hints as what other packages I should uninstall?
[image]

sgalgano, 2019-04-05:
Unless you plan on using the emane Python 3 module on 16.04, there is no reason to install python3-emane. That should alleviate the dependency issues.

enoorani, 2019-04-05:
@sgalgano I was wondering if the problem is caused by not including python3-emane package in the installation process.
I skipped python3-emane package and it ran smoothly 
[image]
when im trying to use emane in core and run it
[image]

I get this error
[image]

I was wondering if the problem is caused by not including python3-emane package in the installation process. The system is using python2.

[image]

also i did "sudo ln -s /usr/local/share/emane /usr/share/emane" based on what has been suggested on the CORE installation guide.

sgalgano, 2019-04-05:
Your issue is unrelated to the exclusion of python3-emane. Your best bet would be to seek assistance from the CORE project.

enoorani, 2019-04-05:
@sgalgano I though it might be an EMANE originated problem since when I use the "emane" command I get "Missing CONFIG_URL"
[image]

sgalgano, 2019-04-05:
See the [emane-tutorial](https://github.com/adjacentlink/emane-tutorial) for more information.

enoorani, 2019-04-08:
@sgalgano @eschreiber-alink Any idea what the problem is? and any suggestions as how to fix it?
running "make deb WITHOUT_PYTHON3=1" yields
[image]

[image]

sgalgano, 2019-04-09:
You may have installed multiple versions of protocol buffers on your build system. The [instructions]( https://github.com/adjacentlink/emane/wiki/Build#ubuntu-14041604-lts) for building emane on 16.04 use protocol buffers 2.

closed 2019-06-07 by sgalgano

legit13, 2020-05-05:
I am also getting an error with EMANE v0.9.3 on Ubuntu 16.04. I got past the 'No ACE Headers' issue, but now I'm here, similar to the thread above.

`tail: cannot open 'emane-0.9.3/debian/changelog' for reading: No such file or directory
dpkg-source: error: tail of emane-0.9.3/debian/changelog gave error exit status 1
dpkg-buildpackage: error: dpkg-source -b emane-0.9.3 gave error exit status 1
Makefile:894: recipe for target 'deb' failed
make: *** [deb] Error 1
`