adjacentlink/emane #102: How do I connect "emane0" from one Virtual Machine (vm) to "emane0" on another Virtual Machine (vm)?

WorksWellWithOthers, 2019-06-06:
How do I connect "emane0" from one Virtual Machine (vm) to "emane0" on another Virtual Machine (vm)?

I currently have EMANE setup on a host machine with 6 NEMs.
I'm trying to replace two of the NEMs with Virtual Machines running on the same host.

VM-1 and VM-2 can ping each other via the Virtual Network Interface Bridge (ens10).

If I launch the LXC containers with ./demo-start I can SSH into Node-1 and ping 10.100.0.2 in each VM.
Outside of the node the VMs can ping each other.

Would using "brctl" to bridge ens10 with emanenode0 be the proper way to go about this?

The other thing I tried was to change"otamanagerdevice" and "eventservicedevice" values to "ens10".
Then launch "sudo emane platform1.xml -r -d -l 3 -f persist/1/var/log/emane.log" in VM-1 and
"sudo emane platform2.xml -r -d -l 3 -f persist/2/var/log/emane.log" in VM-2.
Followed by "brctl addbr test", "brctl addif test ens10 emane0".
Thinking I can "ping 10.100.0.2" from VM-1 afterwards.
However, if I do this I can't ping 10.100.02 or VM-2 from VM-1.

The end goal is to send packets from VM-1 to VM-2 with the EMANE network being in the middle conducting events.

sgalgano, 2019-06-07:
Assuming you want VM-1 and VM-2 acting as hosts hanging off of two radios being emulated with emane in two LXCs:

```
[ VM-1 ] <-- lan1 --> [ Radio LXC 1 (NEM 1) ] <--- OTA --> [ Radio LXC 2 (NEM 2) ] <-- lan2 --> [ VM-2 ]
```

* Add an additional interface in the container configuration of both Radio LXCs to create a lan connection to the VMs.
* On the host system, bridge the host system side of the LXC  lan vif with the host system side of its respective VM lan interface vif. Do this for both lan1 and lan2.
* Add the necessary routes or default route to each VM.

WorksWellWithOthers, 2019-06-07:
To check my understanding of what I need to do:

1. Figure out how to create a lan connection to the VMs using an interface in the container configuration. 
2. Figure out how to bridge the newly created lans with the existing host lan vif. 
3. Figure out how to route to each VM.

It sounds similar to this pdf:
"Both of our EMANE clusters have three virtual
interfaces per virtual machine (eth0, eth1, eth2) each
connected a separate VLAN which is connected to a tap on
the host. The host side of the eth0 taps are added to a
bridge with the physical eth0 interface of the host,
connecting the eth0 interfaces of the virtual machines to
the management network. The virtual eth1 taps are handled
in the same way (being bridged with the host event interface
or bonded channel)."
From Page-3:  https://pdfs.semanticscholar.org/65ff/cad6cfeb8c9cd6e161e7d19ce92466960ed1.pdf

It seems I have a lot of learning to do since I do not yet know how to go about steps 1, 2, and 3.

sgalgano, 2019-06-07:
I just added an additional example ([exp-03](https://github.com/adjacentlink/letce2-tutorial/tree/master/exp-03)) to the [letce2-tutorial](https://github.com/adjacentlink/letce2-tutorial). This example produces the topology you are interested in investigating but uses LXCs in place of VMs for the application hosts. Building and reviewing the configuration may prove helpful.

```
[ host-1 ] <-- lan --> [ node-1 ] <-- OTA --> [ node-2 ] <-- lan --> [ host-2 ]
   .2      10.98.1.0/24    .1                     .1     10.98.2.0/24   .2
```

WorksWellWithOthers, 2019-06-11:
I'm having a difficult time trying to create physical interfaces for each container and setting up the correct routing for them. I do feel like I improved my understanding a bit but I'm not sure if I'm close to solving this or if I still have a long ways to go in terms of learning to figure it out. 

1. From inside [VM-1][Node-1] (10.99.0.1), "ip route add default via 10.99.0.1", does this create LAN1?

Use Case:
ssh [VM-1][Node-1]
ping 10.100.0.3 ([VM-2][Radio-1])

What I tried to build:
10.100.0.1 = [VM-1][Radio-1]
10.100.0.2 = [VM-1][Radio-2]
10.100.0.3 = [VM-2][Radio-1]
10.100.0.4 = [VM-2][Radio-2]

Looking at this:
```
[ host-1 ] <-- lan --> [ node-1 ] <-- OTA --> [ node-2 ] <-- lan --> [ host-2 ]
   .2      10.98.1.0/24    .1                     .1     10.98.2.0/24   .2
```

It looks like my issue may be related to not having my Nodes in the same subnet (not sure how to do this).
My [VM-1][Node-1] is "10.99.0.1" and my [VM-2][Node-1] is also "10.99.0.1".

sgalgano, 2019-06-11:
The 10.99.0.0/16 is a back channel network for controlling the emulation. You should never have to add routes for that.

In exp-03, each radio node has a lan interface for one or more hosts to hang off. The olsr instance running in the radio node advertises this host network. Looking at the routing table of any radio node shows both the OTA addresses 10.100.0/24 of the other radio nodes and the host networks being advertised by those radio nodes.

```
[me@host exp-03]$ ssh node-1
[me@node-1 ~]$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.98.1.0       0.0.0.0         255.255.255.0   U     0      0        0 lan0
10.98.2.0       10.100.0.2      255.255.255.0   UG    2      0        0 emane0
10.99.0.0       0.0.0.0         255.255.255.0   U     0      0        0 backchan0
10.100.0.0      0.0.0.0         255.255.255.0   U     0      0        0 emane0
10.100.0.2      10.100.0.2      255.255.255.255 UGH   2      0        0 emane0
224.0.0.0       0.0.0.0         240.0.0.0       U     0      0        0 bmf0
```

If your desire is to replace the LXC hosts (host-1 and host-2) in exp-03 with VMs, you will need to set the address of one of the interfaces inside each VM to be on its respective radio lan. You will also need to add the host side of the VM's interface to the corresponding linux bridge containing its radio node's lan vif.

For example, exp-03 sets up a bridge for each radio node lan: r1_0 and r2_0. 
```
[me@host ~]$ brctl show
bridge name	bridge id		STP enabled	interfaces
letce0		8000.fe03e17d031d	no		aveth.1.0
							aveth.2.0
							aveth.3.0
							aveth.4.0
r1_0		8000.fe6d372e1fce	no		aveth.1.1
							aveth.4.1
r2_0		8000.fe9139906998	no		aveth.2.1
							aveth.3.1
```
Where, aveth.1.1 is the host side of the radio node node-1's lan0 and aveth.4.1 is the host side of the application-host host-1's lan0; aveth.2.1 is the host side of the radio node node-2's lan0 and aveth.3.1 is the host side of the application-host host-2's lan0.

WorksWellWithOthers, 2019-06-11:
Thank you for your patience.
It seems I was doing things incorrectly by having emane running inside both VM-1 and VM-2.
I will be reattempting everything with emane running outside of VM-1 and VM-2 on the host machine.
I'll update here as I make progress.

WorksWellWithOthers, 2019-06-12:
I've reattempted everything and organized what I did. Doesn't work yet but I'm learning so that's a plus.  

It looks like radio-1 and radio-2 are associated with veth1.2 and veth2.2
I created and added veth1.2 to radio1Bridge and veth2.2 to radio2Bridge using brctl on the host machine.
I then added radio1Bridge to VM-1 as a Virtual Network Interface and it now shows up as ens11.

When I run "ip route get 10.100.0.1" (radio-1) on VM-1 it responds with "10.100.0.1 dev ens10 src 10.100.0.190".
When I run "ip route get" on the host machine it responds with "10.100.0.1 dev virbr0 src 10.100.0.100".
I was surprised that the response from the host machine wasn't "10.100.0.1 dev veth1.2 src 10.100.0.100".

I'm trying to ping radio-1(running inside an LXC on the host machine -- emane-tutorial/0) from VM-1.
The host can't ping radio-1 without first SSHing into Node-1. This is the correct behavior. I'm just bringing this
up to ask if VM-1 will be able to ping radio-1 when the host can't do it without remoting into Node-1.   

I'm wondering if I'm on the right track.
I'm thinking of tinkering with "ip route add" in hopes of things eventually(magic) working.
I'm also having doubts that veth1.2 and veth2.2 are the correct interfaces for me to feed to the VMs.
Should I be looking to somehow bridge emane0 from within Node-1 and make that available to the host then
somehow feed that to the VM?
That's what I thought I did by bridging veth1.2 on the host and making it a NIC on VM-1.

WorksWellWithOthers, 2019-06-13:
Is it possible to add a physical interface to each of the containers through the .xml scripts in the emane-tutorials?

WorksWellWithOthers, 2019-06-13:
Note to self:
To add the LXC container created through emane-tutorial you have to edit the "demo-ctl-host" script called via "demo-start". 

lxc.network.type=veth
lxc.network.name=eth2
...
lxc.network.link = name_of_bridge

WorksWellWithOthers, 2019-06-13:
Is there a difference between adding VETH1.2 (container vif) to the bridge (myBridge) via lxc.conf versus doing it after the container creation using "brctl addif myBridge VETH1.2"?

WorksWellWithOthers, 2019-06-15:
I explored creating another interface on the LXC containers. 
I'm thinking once I can establish a connection between an interface on the LXC container with an interface on my VM I can work on next steps to replacing the container with the VM.

I was able to successfully have the democtl-host create another VETH (VETH1.3). 
Then I realized I have to actually remote into Node-1 then launch Emane to get a NEM (radio).
ETH1 and ETH2 are the LXC interfaces. Which means I didn't create the interfaces in the LXC containers like I thought I did. 

Questions:
1. What exactly is VETH?
2. How is EMANE0 created? (I'm relating EMANE0 with a Radio)
3. Do I need to know these details if I am attempting to replace one of the Radios with a VM? (Making sure I'm not going the wrong direction)
4. Is there a difference between adding VETH1.2 (container vif) to the bridge (myBridge) via lxc.conf versus doing it after the container creation using "brctl addif myBridge VETH1.2"?

Here are some other things I discovered and I am confused about...

10.99.0.100 -- Emanenode0 is a bridge containing all the node (backchannels).
10.99.0.1 -- You have to go into one of these nodes to create a Radio (NEM / emane0) by launching Emane.
10.100.0.1 -- emane0 is the Radio created by launching emane platform.xml

I think 10.99.0.100 can also be used as the Host IP. 
5. What is the difference between the Host-IP and 10.99.0.100?

sgalgano, 2019-06-15:
There is a chance things have become more complicated than they need to be. If your goal is to setup an experiment using VMs instead of containers, you don't need any of this. If you are using a single VM instance as an emulated radio, all you need is to start multiple VMs with your virtualizer and configure them with a network they can communicate over. As long as you have connectivity between them, you can configure and run emane on each. If you look at [emane-tutorial 0](https://github.com/adjacentlink/emane-tutorial/wiki/Demonstration-0), you will find examples of the configuration and commands you need to run in each VM.

If your goal is to mix VM instances and LXCs in the same experiment using the same server, you have to deal with the items we have been discussing in this thread.

My suggestion would be to stick with either all VMs or all LXCs until you have a better handle on virtualization plumbing.

WorksWellWithOthers, 2019-06-17:
I'm now using VMs only to get a better handle on the virtualization plumbing.
I'm assuming I have to launch emane without ./demo-start?

sgalgano, 2019-06-17:
The [emane-tutorial](https://github.com/adjacentlink/emane-tutorial) is designed to give you 10 nodes with network stack isolation (using LXCs) to walk through simple emane experiments. Don't confuse the simple [demo test flow](https://github.com/adjacentlink/emane-tutorial/wiki#tutorial-test-flow) with emane. emane is just an application (see [FAQ](https://github.com/adjacentlink/emane/wiki/FAQ#does-emane-require-virtualization)).

How you run your experiments and control them is outside the scope of the emulator. We have some open source tools that do test control with varying degrees of formality ([etce](https://github.com/adjacentlink/python-etce), [letce2](https://github.com/adjacentlink/letce2)), if you are looking for something.

WorksWellWithOthers, 2019-06-22:
I made some progress and I can now ping 10.100.0.6 from Node-3 and ping 10.100.0.3 from the VM through the 10.10.7.X veths.

1. Is there a way to ping each other through the NEMs instead?
2. Should I change the NEMs IP to 10.10.7.X via platform.xml?

LXC Node-3
emane platform3.xml -r -d -l 3 -f persist/3/var/log/emane.log
ip route add 10.100.0.6 dev ETH3

KVM 
emane platform6.xml -r -d -l 6 -f persist/6/var/log/emane.log
ip route add 10.100.0.3 dev ENS11

Host
brctl addif virbr0 veth3.3

The veths are on 10.10.7.X

PathlossEvent is done through Python:
event = PathlossEvent()
event.append(3,forward=10)
event.append(6,forward=10)
service.publish(3,event)
service.publish(6,event)

WorksWellWithOthers, 2019-06-24:
I suspect my challenges are now related to:
<param name="otamanagerdevice" value="eth1"/>
<param name="eventservicedevice" value="eth1"/>

I have not been adjusting these at all.

VM-1:
emane platform1.xml -r -d -l 3 -f persist/1/var/log/emane.log

VM-2:
emane platform2.xml -r -d -l 3 -f persist/2/var/log/emane.log

VM-?:
emaneevent-pathloss 1:2 90 -i <not sure which interface>

WorksWellWithOthers, 2019-06-25:
Thank you for the help. I now have a VM acting as a host hanging off of radios being emulated with emane. I now want to try and run emane without the demo. I followed the steps from demo_0 and demo_1 without the demo-start. I'm not sure what I'm doing wrong, the machines can communicate on their default interfaces. I can have them communicate with emane0 if I route it through the default interfaces (ip route add 10.100.0.2 dev 10.10.7.2). I can't seem to get pathloss to affect their communications.  

Machine-1:
emane platform1.xml -r -d -l 3 -f persist/1/var/log/emane.log

Machine-2:
emane platform2.xml -r -d -l 3 -f persist/2/var/log/emane.log

Unsure which machine to run pathloss from -- I do it on both:
emaneevent-pathloss 1:2 90 -i not_sure_which_interface -- I use the one I can ping the other machine with.

**1. Do the NEMs have to be on the same subnet that the machines are pinging each other with?
2. Should I change the NEMs IP to 10.10.7.X via platform.xml?**

I suspect my challenges might be related to figuring out how to properly configure the otamanager and eventservice on both machines.
```
<param name="otamanagerdevice" value="eth1"/>
<param name="eventservicedevice" value="eth1"/>
```

closed 2019-07-01 by WorksWellWithOthers