adjacentlink/emane #103: Question related to EMANE time management

omidnabi, 2019-06-07:
Hi,
 I just have a question about EMANE way of handling time. Is it possible to add time management into EMANE? like if I'm using EMANE in  a simulation with other models that are time managed, would I be able to apply that time management to the EMANE, so it would run at the same rate as the other models in the simulation?

Thank you!

sgalgano, 2019-06-07:
EMANE runs in real time. Moving EMANE (emulator + radio models) from emulation to simulation, would be a complex change.

omidnabi, 2019-06-07:
Let's say I'm willing to tackle this complex change, could you please point me in the right direction? I'm trying to scope this task and I need to figure out where I need to start from and what components of the code I need to consider for a redesign in order to implement the ability to step through time. 

Thank you in advance for your time and guidance!

closed 2019-11-07 by sgalgano