adjacentlink/emane #106: How to implement a new event class in EMANE v1.2

ChengDaHaI, 2019-07-05:
Recently, I am focus on integrating the EMANE network emulator with our network optimization platform. In the work, we need to adjust the TxPower of each node dynamically. It seems that we have to using event service to adjust the txpower after the network is emulated. 

However, i don't find anything about how to implement a new event in the latest EMANE(v1.2) while this part is included in the v0.7.1 developer's guide. Therefore, i wonder if we could implement a new event in the latest EMANE or not. If we can, could you share some hints about that?

Thanks!

sgalgano, 2019-07-06:
I recommend taking a look at the [control port](https://github.com/adjacentlink/emane-control-port-tutorial) interface. This is the mechanism we use to dynamically control running-state modifiable configuration items such as [txpower](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#txpower).

ChengDaHaI, 2019-07-09:
@sgalgano Thank you!

closed 2019-10-02 by sgalgano