adjacentlink/emane #113: Can we  combine EMANE with GNU radio?

ChengDaHaI, 2019-09-26:
In the introduction, it says "can be combined with existing Software Defined Radio (SDR) implementations to enable shared code emulation". Does that mean EMANE can be combined with GNU Radio or other similar software? If it is possible, could you share hints about how to do that? 

Best,

closed 2019-10-01 by ChengDaHaI

sgalgano, 2019-10-02:
[srsLTE-emane](https://github.com/adjacentlink/srsLTE-emane) is an open source example of using emane with an SDR. The basic idea is to find the level in the SDR stack where you can create an interface to convert from SDR APIs to emane. Usually this occurs in or around the SDR's MAC layer.

GNU Radio focuses on the parts of a waveform that are abstracted out in emane (e.g., sampling, bit-packing, modulation). In the srsLTE-emane example, low layer physical layer functionality was replaced by emane with the rest of the upper layers remaining the same.

ChengDaHaI, 2019-10-02:
> [srsLTE-emane](https://github.com/adjacentlink/srsLTE-emane) is an open source example of using emane with an SDR. The basic idea is to find the level in the SDR stack where you can create an [...]

So, from my understanding, given a gnuradio+usrp project, we can replace the original physical layer by EMANE while keep the upper layer (almost) unchanged if we can find or create an interface (Modem hardware abstraction layer?), am i right? Even more, since EMANE simulates a MAC layer, can the original MAC layer be replaced by EMANE ?

Thank you!