adjacentlink/emane #114: Packet data changes in EMANE

szl0144, 2019-10-01:
I used MGEN to change the packet data and TOS in IPv4 header at first, however. EMANE doesn't run correctly after I changed these values in MGEN, message arrival rate, message length, etc, become some default values, the system doesn't run as I want. 

Then I started to change them in EMANE, I modify the data value at the pkt data's address and change the TOS in IPv4 pkt header, but I meet the same bugs again when I run the MGEN, message arrival rate, message length become some default value.  Can you tell me if it is available to change the packet data in MGEN?

sgalgano, 2019-10-02:
We use mgen with emane and do not have any issue setting flow parameters. Neither emane nor the radio models contained in the mainline distribution change the contents of any packet.

You need to provide more information detailing the emulator and radio model configuration and version, and how to reproduce the issue you are seeing.

szl0144, 2019-10-04:
Thanks for your suggestion. The model is TDMA, EMANE's version is 1.2.4, and MGEN's version is 5.02.

I generate the traffic flow using MGEN and integrate pkts into original version EMANE, this time EMANE runs correctly without the previous bugs, but the TOS in IPv4 header can't be changed and always be 0. Does EMANE support changes in TOS part?

I also try to extract the message data payload from pkt, I shift the packet buffer's pointer from IPv4 header's address to UDP header's, next shift to the message data's address in function readDevice() which is in file virtualtransport.cc, but the value extracted from the data part is not what I set. Because I set the data in MGEN as 8 bits data, I set the data payload variable type as uint8_t to print the data out. I am not sure if my way to get the data payload is right.

Sincerely,
Shaoyi

sgalgano, 2019-10-07:
EMANE *supports* IP TOS, in that it respects the TOS value set by an application. Some models, including the TDMA model, use TOS to map to priority queues. Note that with TDMA, your [schedule]( https://github.com/adjacentlink/emane/wiki/TDMA-Model#defining-a-tdma-schedule) will determine if/how different TOS values are handled.  You most likely want all your slot `class` attributes to be set to 3 and have the [queue.strictdequeueenable](https://github.com/adjacentlink/emane/wiki/TDMA-Model#queuestrictdequeueenable) configuration parameter set to `no`.

As far as looking at packet payloads within the virtual transport, keep in mind that you are going to have an ethernet header in front of your IP packet.

szl0144, 2019-10-11:
I see, thanks for your explanation! I have extracted the data part from the pkt buffer, 

I noticed there are two kind of data in MGEN user manual, the first one is message data payload in **[6.11. DATA](https://downloads.pf.itd.nrl.navy.mil/docs/mgen/mgen.html#d0e1860)**, and second part is MGEN message payload in **[MGEN Message Payload](https://downloads.pf.itd.nrl.navy.mil/docs/mgen/mgen.html#d0e1860)**, the data payload part is included in MGEN message payload, Does it mean I need to shift the pointer 4*13bytes from UDPHeader's address to the location of data payload and get the value saved in it?

closed 2019-11-07 by szl0144