adjacentlink/emane #115: > [srsLTE-emane](https://github.com/adjacentlink/srsLTE-emane) is an open source example of using emane with an SDR. The basic idea is to find the level in the SDR stack where you can create an interface to convert from SDR APIs to emane. Usually this occurs in or around the SDR's MAC layer.

ChengDaHaI, 2019-10-03:
> [srsLTE-emane](https://github.com/adjacentlink/srsLTE-emane) is an open source example of using emane with an SDR. The basic idea is to find the level in the SDR stack where you can create an interface to convert from SDR APIs to emane. Usually this occurs in or around the SDR's MAC layer.
> 
> GNU Radio focuses on the parts of a waveform that are abstracted out in emane (e.g., sampling, bit-packing, modulation). In the srsLTE-emane example, low layer physical layer functionality was replaced by emane with the rest of the upper layers remaining the same.

So, from my understanding, given a gnuradio+usrp project, we can replace the original physical layer by EMANE while keep the upper layer (almost) unchanged if we can find or create an interface (Modem hardware abstraction layer?), am i right? Even more, since EMANE simulates a MAC layer, can the original MAC layer be replaced by EMANE ?

Thank you!

_Originally posted by @ChengDaHaI in https://github.com/adjacentlink/emane/issues/113#issuecomment-537726245_

sgalgano, 2019-10-03:
Your understanding pertaining to creating an MHAL interface with the aim of keeping the upper waveform layers the same/minimally-changed is correct. EMANE does not emulate a MAC layer. You will need to develop some level of MAC functionality. The complexity of which depends on how the original waveform was designed and implemented.

ChengDaHaI, 2019-10-03:
> Your understanding pertaining to creating an MHAL interface with the aim of keeping the upper waveform layers the same/minimally-changed is correct. EMANE does not emulate a MAC layer. You will need [...]

Thank you very much for the elaboration.  However, i don't understand why you say "EMANE does not emulate a MAC layer". In the platform.xml config file, there is a part in radio model config file(***nem.xml) defining a mac layer. If that doesn't account for the MAC layer, where do we get the mac layer?

sgalgano, 2019-10-03:
You have to write a MAC plugin. That XML is instructing the emulator as to which plugin to load and how to configure it.

closed 2019-11-07 by sgalgano