adjacentlink/emane #116: EMANE Olsrd Demo with separate EMANE control and OTA interfaces

hideogump, 2019-10-04:
I am extending the EMANE olsrd demo 0 with the goal of separating out the control and OTA interfaces to separate Linux bridge devices that are created by the host scripts.
This will allow me to more easily inspect the network packets flowing over the OTA interface rather than separating out by IP.

To achieve this goal, I have defined a separate virtual interface in each of the container LXC configuration files similar to the folowing as eth2. This container also contains an eth0 as the host interface, and eth1 as the dedicated EMANE SSH control interface. The parent host script creates the Linux bridges that are linked to within the container.

The problem is that when EMANE runs, it creates a TAP interface as "emane0" whose IP collides with eth2 created within the container.

The following are some configuration file samples.
It would appear my understanding of the transports needs some clarity.

How might this separation of virtual interfaces be accomplished?

My longer term goal is to have olsrd running within an EMANE container bridging two logically separate network.

-----------------------------------

LXC Configuration Excerpt:

# ota interface
lxc.network.type=veth
lxc.network.flags=up
lxc.network.hwaddr=00:00:00:01:01:02
lxc.network.ipv4=10.100.0.1/24
lxc.network.name=eth2
lxc.network.veth.pair=veth.ota.001
lxc.network.link=ota

# br.ctl interface
lxc.network.type=veth
lxc.network.flags=up
lxc.network.hwaddr=00:00:00:01:01:01
lxc.network.ipv4=10.99.0.1/24
lxc.network.name=eth1
lxc.network.veth.pair=veth.ctl.001
lxc.network.link=br.ctl

# radiohost1 interface
lxc.network.type=veth
lxc.network.flags=up
lxc.network.hwaddr=00:00:00:01:01:00
lxc.network.ipv4=135.1.1.1/16
lxc.network.name=eth0
lxc.network.veth.pair=veth.host.001
lxc.network.link=host1

-------------------------------

PLATFORM.XML

  param name="otamanagerchannelenable" value="on"/>
  param name="otamanagerdevice" value="eth2"/>
  param name="otamanagergroup" value="224.1.2.8:45702"/>
  param name="eventservicegroup" value="224.1.2.8:45703"/>
  param name="eventservicedevice" value="eth1"/>
  param name="controlportendpoint" value="0.0.0.0:47000"/>
  nem id="1" definition="nem.xml">
    transport definition="transvirtual.xml">
      param name="address" value="10.100.0.1"/>
      param name="mask" value="255.255.255.0"/>
    /transport>
  /nem>
------------

NEM.XML

!DOCTYPE nem SYSTEM "file:///usr/share/emane/dtd/nem.dtd">
nem>
  transport definition="transvirtual.xml"/>
  mac definition="mac.xml"/>
  phy definition="phy.xml"/>
/nem>

----

MAC.XML

!DOCTYPE mac SYSTEM "file:///usr/share/emane/dtd/mac.dtd">
mac library="bypassmaclayer"/>

sgalgano, 2019-10-04:
You either need to change the network used in your `lxc.conf` for `veth.ota.001`:

```
lxc.network.ipv4=10.100.0.1/24
lxc.network.name=eth2
lxc.network.veth.pair=veth.ota.001
```

or the network you are using in your `platform.xml`: 

```xml
  <nem id="1" definition="bypassnem.xml">
    <transport definition="transvirtual.xml">
      <param name="address" value="10.100.0.1"/>
      <param name="mask" value="255.255.255.0"/>
    </transport>
    </nem>
</platform>
```

They can't be the same.

The transport *address* is the radio over-the-air network corresponding to `emane0`.  This is the boundary when using the Virtual Transport, how packets enter and exit (routed to/from) emulation space.

The `veth.ota.001` is an isolated *control network* that multicast OTA messages are transmitted on. This is your emulated RF signaling. Think of it as emulator-to-emulator messaging.

hideogump, 2019-10-07:
Thanks for the quick response, I changed the LXC configuration to lxc.network.ipv4=10.98.0.1/24, and all is well.

closed 2019-10-07 by hideogump