adjacentlink/emane #117: Host to Host Traffic Routing from Host Container to Radio Container to OTA Interface

hideogump, 2019-10-05:
This is likely a Linux network routing issue, not EMANE issue proper, but I wanted to obtain the best practices guidance when using EMANE in a network configuration consisting of two containers for a logical "node":

1. A host container using a class C IP address as eth0 (e.g. 135.1.1.2), which connects to 
2. A radio container (e.g. 135.1.1.1) running EMANE using the standard virtual transport IP of 10.100.0.x for the interface emane0.

I want to send host traffic from say host #1 (135.1.1.2) to host #2 (135.1.2.2), but it is not currently doing so across the EMANE OTA network using some default routes in the radio container.

Olsrd routing is running in each of the radio containers, and they are able to find the other radio interfaces of a peer (i.e. IP addresses in the OTA 10.100.0.0/24 network), and a dump of the IP routing table on the radio container running EMANE indicates OTA connectivity between the radio container's EMANE instances.

What may be missing is the routing necessary in the radio container to forward traffic from the host container (received on interface eth0) to the EMANE virtual transport (on interface emane0).  The host traffic uses the IP convention of 135.1.x.2, and I would want this host traffic to be carried across the network intact (perhaps without any form of NATing) to the destination host.

In the radio container, I have the default route set to the emane0 with the Gateway set to the IP assigned to the radio (e.g. 10.100.0.1).

I would like the solution to be somewhat generic and not specify specify host IPs as part of static routing if possible, so that it is host IP agnostic.

I was considering tying the two interface (host eth0 and OTA emane0) in the radio container to forward traffic between them using IPTABLES, coupled with enabling routing in the kernel, something like the following:

net.ipv4.ip_forward = 1
sudo iptables -t nat -A POSTROUTING --out-interface eth1 -j MASQUERADE  
sudo iptables -A FORWARD --in-interface eth0 -j ACCEPT

What is the recommended solution for what is probably a very common test modality?

sgalgano, 2019-10-07:
For our emane with olsrd with application-host containers hanging off of radio LXC lans, we use olsr HNAs (Host Network Advertisement). This is what is done in [letce2-tutorial/exp-03](https://github.com/adjacentlink/letce2-tutorial/tree/master/exp-03). I believe this would address your needs.

hideogump, 2019-10-08:
Thanks Steve, for the quick response. I was able to gain some insight from the referenced example, and now have HNA enabled for olsrd and the routing tables for all radio nodes correctly identify the host network attached to the radios (135.1.x.2 for the host, 135.1.x.1 for the radio data).

However, I am still unable to PING host to host.  What I observe is the ARP initiation from say host 1 on eth0 to eth0 on radio1, which as a broadcast message would presumably need to be routed to each OTA radio on the OTA network.  The ARPs do not seem to do so. This may be question that is broadly associated with how EMANE can be configured to handle host broadcast types of traffic.   What does EMANE require for broadcasting inbound ARP traffic from the host to the OTA network in an environment which includes olsrd with the Basic Multicast Forwarding plug-in.  Is it require to simply populate the ARP entries in the host container as an interim solution?

The following are some terminal snippets where I am using the 10.100 network for the EMANE transport, 10.98 for the isolation OTA interface,  the 135.1 network as the host interface, and 10.99 for the isolated control network.

-------
4 Node Demo Radio-001 ROUTING TABLE

[rrpucsuser@radio-001 ~]$ route -n
Kernel IP routing table
Destination Gateway Genmask Flags Metric Ref Use Iface
10.98.0.0 0.0.0.0 255.255.255.0 U 0 0 0 eth2
10.99.0.0 0.0.0.0 255.255.255.0 U 0 0 0 eth1
10.100.0.0 0.0.0.0 255.255.255.0 U 0 0 0 emane0
10.100.0.2 10.100.0.2 255.255.255.255 UGH 2 0 0 emane0
10.100.0.3 10.100.0.3 255.255.255.255 UGH 2 0 0 emane0
10.100.0.4 10.100.0.4 255.255.255.255 UGH 2 0 0 emane0
135.1.1.0 0.0.0.0 255.255.255.0 U 0 0 0 eth0
135.1.2.0 10.100.0.2 255.255.255.0 UG 2 0 0 emane0
135.1.3.0 10.100.0.3 255.255.255.0 UG 2 0 0 emane0
135.1.4.0 10.100.0.4 255.255.255.0 UG 2 0 0 emane0
224.0.0.0 0.0.0.0 240.0.0.0 U 0 0 0 bmf0

[rrpucsuser@radio-001 ~]$ ping -R radiohost2data
PING radiohost2data (135.1.2.2) 56(124) bytes of data.
64 bytes from radiohost2data (135.1.2.2): icmp_seq=1 ttl=63 time=0.306 ms
RR:     radio1ota (10.100.0.1)
    radio2data (135.1.2.1)
    radiohost2data (135.1.2.2)
    radiohost2data (135.1.2.2)
    radio2ota (10.100.0.2)
    radio1ota (10.100.0.1)

[rrpucsuser@radio-002 ~]$ arp
Address                  HWtype  HWaddress           Flags Mask            Iface
radio1ota                ether   02:02:00:00:00:01   C                     emane0
radiocontrol             ether   fe:7c:6e:51:5a:fa   C                     eth1
radiohost2data           ether   00:00:00:02:04:00   C                     eth0

[rrpucsuser@radio-002 ~]$ ping radiohost1data -R -c 1
PING radiohost1data (135.1.1.2) 56(124) bytes of data.
64 bytes from radiohost1data (135.1.1.2): icmp_seq=1 ttl=63 time=0.258 ms
RR:     radio2ota (10.100.0.2)
    radio1data (135.1.1.1)
    radiohost1data (135.1.1.2)
    radiohost1data (135.1.1.2)
    radio1ota (10.100.0.1)
    radio2ota (10.100.0.2)

--- radiohost1data ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.258/0.258/0.258/0.000 ms

[rrpucsuser@radiohost-001 ~]$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         135.1.1.1       0.0.0.0         UG    0      0        0 eth0
10.99.0.0       0.0.0.0         255.255.255.0   U     0      0        0 eth1
135.1.0.0       0.0.0.0         255.255.0.0     U     0      0        0 eth0

sgalgano, 2019-10-08:
If you have not already done so, it would be beneficial to run [letce2-tutorial/exp-03](https://github.com/adjacentlink/letce2-tutorial/tree/master/exp-03) and interact with the running experiment. Host-to-host communication works fine. 

```
[me@host-1 ~]$ ping -R 10.98.2.2 -c 1
PING 10.98.2.2 (10.98.2.2) 56(124) bytes of data.
64 bytes from 10.98.2.2: icmp_seq=1 ttl=62 time=5.49 ms
RR: 	10.98.1.2
	10.100.0.1
	10.98.2.1
	10.98.2.2
	10.98.2.2
	10.100.0.2
	10.98.1.1
	10.98.1.2

--- 10.98.2.2 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 5.488/5.488/5.488/0.000 ms

[me@host-1 ~]$ traceroute -n 10.98.2.2
traceroute to 10.98.2.2 (10.98.2.2), 30 hops max, 60 byte packets
 1  10.98.1.1  0.064 ms  0.027 ms  0.023 ms
 2  10.100.0.2  3.718 ms  5.446 ms  7.245 ms
 3  10.98.2.2  8.507 ms  10.071 ms  11.726 ms
```

For your setup, verify ip forwarding is enabled in the radio containers:

```
[me@node-1 ~]$ sysctl net.ipv4.ip_forward
net.ipv4.ip_forward = 1
```

hideogump, 2019-10-14:
Hi Steve, in case you have any other insights, here is an update.

IP forwarding is set correctly and verified within each container.  

Using an olsrd.conf file that is nearly identical to the examples, and an IP convention that now matches the demo so that we can communicate using the same IP "currency", am not able to PING host to host (which is host1data to host2data using the attached hosts file).  

I am able to PING from radio node to the host data interface of the other radio (e.g. from radio2ctl to host1data across the EMANE OTA network, or radio1ctl to host2data).

[rrpucsuser@radio-002 ~]$ ping -R radio1data
PING radio1data (10.98.1.1) 56(124) bytes of data.
64 bytes from radio1data (10.98.1.1): icmp_seq=1 ttl=64 time=0.836 ms
RR: 	radio2emane (10.100.0.2)
	radio1data (10.98.1.1)
	radio1data (10.98.1.1)
	radio2emane (10.100.0.2)

The routing tables look correct for the hops in the radio nodes showing the HNA elements defining the upstream host IPs.  For example, for the radio 2 container for a 2 node configuration:

[rrpucsuser@radio-002 ~]$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.98.1.0       10.100.0.1      255.255.255.0   UG    2      0        0 emane0
10.98.2.0       0.0.0.0         255.255.255.0   U     0      0        0 eth0
10.99.0.0       0.0.0.0         255.255.255.0   U     0      0        0 eth1
10.100.0.0      0.0.0.0         255.255.255.0   U     0      0        0 emane0
10.100.0.1      10.100.0.1      255.255.255.255 UGH   2      0        0 emane0
10.101.0.0      0.0.0.0         255.255.255.0   U     0      0        0 eth2
224.0.0.0       0.0.0.0         240.0.0.0       U     0      0        0 bmf0

What I observe in the radio container itself is ARPs coming in from the host container (e.g. host1ctl, or host2ctl), but these ARPs are not be forwarded to the emane0 emulation boundary from what I can tell.  

I do see OLSRD traffic on the emane0 interface (as well as the OTA interface as multicast) that is occurring at my defined HELLO intervals from the olsrd.conf file.

When running the olsrd command manually in the radio container, the two plugins seem to load without any event or error.   When I use tshark on the Broadcast Multicast Forwarding (BMF) olsrd plugin, it shows no activity on that particular interface.

Attached are some terminal dumps to accompany this description.

----------------------

[rrpucsuser@radio-002 ~]$ ps -ax
  PID TTY      STAT   TIME COMMAND
    1 pts/0    S+     0:00 /usr/sbin/init.lxc --name radio-002 --lxcpath /var/lib/lxc --logpriority
   17 ?        Ss     0:00 /usr/sbin/sshd -o PidFile=/home/rrpucsuser/emane/rrpucs/tests/baseline/p
   26 ?        Ssl    0:00 emane /home/rrpucsuser/emane/rrpucs/tests/baseline/config/rrpucs/node-00
   40 pts/0    S+     0:00 python -u /usr/sbin/redhawk.py
   42 ?        Ss     0:00 /usr/local/sbin/olsrd -f /home/rrpucsuser/emane/rrpucs/tests/baseline/co
   50 ?        Ss     0:00 sshd: rrpucsuser [priv]
   55 ?        S      0:00 sshd: rrpucsuser@pts/0
   56 pts/0    Ss     0:00 -bash

sudo /usr/local/sbin/olsrd -f /home/rrpucsuser/emane/rrpucs/tests/baseline/config/rrpucs/node-002/olsrd.conf -d 2
sudo /usr/local/sbin/olsrd -f /home/rrpucsuser/emane/rrpucs/tests/baseline/config/rrpucs/node-002/olsrd.conf -d 1

 *** olsr.org - pre-0.9.8-git_0000000-hash_dce3d5473b0637a358df3777b5012b75 ***
 http://www.olsr.org

Parsing file: "/home/rrpucsuser/emane/rrpucs/tests/baseline/config/rrpucs/node-002/olsrd.conf"
Debug level: 0
IpVersion: 4
Lock file /home/rrpucsuser/emane/rrpucs/tests/baseline/persist/node-002/radio-002/var/run/olsrd.lock
Clear screen enabled
Link quality aging factor 0.200000
HNA IPv4 entry: 10.98.2.0/255.255.255.0
setting ifs_in_curr_cfg = 0
	HELLO interval: 5.00
	HELLO validity: 20.00
Plugin: olsrd_txtinfo.so.1.1
Plugin param key:"accept" val: "0.0.0.0"
Plugin: olsrd_bmf.so.1.7.0
Plugin param key:"DoLocalBroadcast" val: "no"
[... 454 lines cut ...]
10.98.2.1 radio2data
10.98.2.2 host2data
10.98.3.1 radio3data
10.98.3.2 host3data
10.98.4.1 radio4data
10.98.4.2 host4data

# CAM Interfaces
10.97.0.1 cam1
10.97.0.2 cam2
10.97.0.3 cam3
10.97.0.4 cam4

# Radio EMANE Transport Interfaces
10.100.0.200 radioemane
10.100.0.1 radio1emane
10.100.0.2 radio2emane
10.100.0.3 radio3emane
10.100.0.4 radio4emane

# Radio EMANE OTA Interfaces
10.101.0.1 radio1ota
10.101.0.2 radio2ota
10.101.0.3 radio3ota
10.101.0.4 radio4ota

sgalgano, 2019-10-14:
Did you run [letce2-tutorial/exp-03](https://github.com/adjacentlink/letce2-tutorial/tree/master/exp-03) and verify host-to-host communication?

Do you have a default route on each host pointing to its respective radio?

hideogump, 2019-10-15:
Yes, I was able to get the letce2 tutorial to work.  There is a default route in each host container, and what I have observed is the ARP broadcast message which precedes any traffic from host to host, is observed on the eth0 (host lan) interface to the radio container, but is not getting routed to the emane0 emulation/transport boundary.  It was my assumption, perhaps incorrectly, that the Broadcast Multicast Flooding plugin to the OLSRD application might be responsible for this activity.  Perhaps you have some thoughts on how initial broadcast traffic, such as ARPs, would nominally be routed to the EMANE interface for transmission over the air.

sgalgano, 2019-10-15:
The host containers are arping for the MAC address of the IP next hop for a respective destination based on the routing table. In this case, the attached radio is the default route out, so it is the target of the host container's arp. If you are not using default routes, you would need specific host network routes on each host container - but those would still point to the host container's attached radio and work the same way. The only arps going over-the-air are the ones generated by the radio container looking for the the MAC address of the IP next hop for self-generated or IP forwarded traffic. ARPs are not forwarded.

hideogump, 2019-10-15:
Thanks for the input.  I was working with the assumption that the models with paired radio and host containers acted somewhat as a Layer 2 Ethernet switch or device, and as such host to host ARP requests (using broadcast destination) and ARP responses (using unicast destination) would actually traverse the entire OTA network intact to the actual host containers destinations.   That would be similar to the Layer 2 radios we design.

sgalgano, 2019-10-15:
We have done custom transports for customers using [OVS](https://www.openvswitch.org) for that purpose. You might want to start there.

hideogump, 2019-10-15:
Thanks Steve for focusing my attention into the right direction.  I was making some false assumptions about OTA ARPs host to host.  I now have it working properly based on your input.  Closing this baby and long thread out.

closed 2019-10-15 by hideogump