adjacentlink/emane #119: Specifying Transmission Durations

hideogump, 2019-10-15:
I currently have a NEM stack based on the pass through MAC and PHY, where the time of transmission is based on the arrival of a packet at the EMANE MAC layer, that is to say, the MAC behavior is implemented external to the EMANE stack based on time of arrival of the network pakcet.  

What settings are available in perhaps the PHY layer (or specified in a particular PHY layer model other than the passthrough) for controlling the transmit duration (which in practice would be calculated by modulation mode, etc.)?   Does a transmission control message need to be generated to control this behavior in say a custom shim (or model)?

sgalgano, 2019-10-15:
You will need to send a [FrequencyControlMessage](https://adjacentlink.com/documentation/emane/v1.2.1/EmulatorPhysicalLayer.html#FrequencyControlMessage) along with the downstream packet. The FrequencyControlMessage contains one or more frequency segments, each having a duration.

hideogump, 2019-10-15:
Thanks Steve, I have written some custom shims, so that might be a good place to implement this customization in say a shim that sits above the MAC layer.  In the absence of this control message, is there a default transmit duration using the simple pass through MAC and PHY models?

sgalgano, 2019-10-15:
The bypass phy has little to no logic in it. It [zeros out most of the common phy header fields](https://github.com/adjacentlink/emane/blob/master/src/models/phy/bypass/phylayer.cc#L69)  and does not process any control messages.

closed 2019-10-15 by hideogump