adjacentlink/emane #121: Maximum Data Transfer Rates Host to Host with Bypass MAC and PHY

hideogump, 2019-10-16:
I have a 4 container demo with 2 host containers handing off of 2 radio containers, both of which have the simple bypass MAC and PHY modules.  I am attempting to determine the highest data rate that can be passed host to host using the standard IPerf3 application running in each host container, one the server, and the other the client.  

I am monitoring both the the processor loading of the host computer system (not to be confused with host container) and each of the two radio containers (using 'top' or 'htop').    The host computer systems in a quad core high performance Intel NUC Skull Canyon with hyper threading so that 8 cores are available to the emulation environment.

What I am observing is that I can drive (somewhat unidirectional since Iperf is client - server) with 0% dropouts up to around 10 Mbps.  For higher rates, packets are dropped, but I see somewhat a flat dropout rate about say 50 Mbps of around 5 percent.

CPU loading in both the radio containers and the host system have at least 50 percent margin even at data rates of 100 Mbps, so I do not believe the system is processor bound.  At 10 Mbps, the CPU loading in the radio container is around 7 percent.

I was expecting perhaps a higher data rate for this somewhat stripped down EMANE NEM stack based on the strength of the host system itself in terms of resources.

I also do not believe looking through the bypass PHY layer source code that packet dropping is occurring in the receiver, nor in the MAC layer.  

I have attempted to dump using emansh the PHY layer tables, which uses the common layer statistics, without any success -- perhaps my entry commands are not correct.

So the question then is what is the upper bound on data transfer rates that one might expect for this type of NEM configuration which has no observed platform CPU or memory limitations?

hideogump, 2019-10-18:
Note that I was able to get some significant improvements in maximum data rate by ensuring that the logging levels for all EMANE components was set to ERROR level. I am able to get around 36 Mbps of traffic with no errors, and the EMANE loading in a radio container is around 13 percent. For the 100 Mbps data rate, I get a drop percentage of less than 2 percent with EMANE CPU loading around 20 percent.

In the drop cases, none of the interfaces in the radio container via ifconfig report any dropped packets, so the drop location needs to be researched a bit more.

I have had experiences in the past where the host itself running Iperf is the actual source of test errors due to CPU limitations.  I am currently running IPerf using a nice level of -20 to possibly enhance the performance of Iperf in the host container.

Does anyone have any experience is attempting to run higher data rates through EMANE?

hideogump, 2019-10-19:
Using the NutTcp test application, I am able to get 100 Mbps data rates with no data losses.  Relative to the time distribution of Iperf, which is very bursty, NutTcp uses a time distribution of packets proportional to the specified user data rate.  Since the radio containers are not showing any ifconfig packet drops, I am not yet convinced that the data drops from  Iperf3 testing in the host containers are not a function of execution of the Iperf binary in the host containers, certainly an experience I have had with hardware implementations at higher data rates.  I would be interested in what the community has experienced for similar maximum throughput test ambitions in EMANE.

sgalgano, 2019-10-20:
Here are some tips for maximizing emulation performance.

On your emulation servers:
* Do not run X
* Disable all unnecessary services
* If running `tuned`, set the active profile to `latency-performance`
* If recording application data (e.g., mgen) do so to ramdisk

LXC configuration:
* Uniquely assign 1 or more cores to each radio container (depending on model resource needs)
* Limit radio container processes to what would be run on the radio: emane, shared-code waveform stack, routing protocols, etc.
* Uniquely assign 1 or more cores to application-host containers hanging off of radio container LAN interfaces. Generate application traffic in these containers.

hideogump, 2019-10-21:
Thanks Steve for suggesting options available to the broader EMANE community for performance tuning.  It is probably not a common issue when dealing with the types of data rates experienced over modelling RF media, but with data rates and hardware always trending to more and more, it will apply more broadly over time.

closed 2019-10-21 by hideogump