adjacentlink/emane #122: The "datarate" parameter in RFpipe model

ChengDaHaI, 2019-10-18:
Just wonder how can you set the transmit [datarate](https://github.com/adjacentlink/emane/wiki/RF-Pipe-Model#datarate) in MAC layer? It says it is used to compute the transmit delay (packet size/datarate). 

However, from my understanding, the transmit data rate should be related to the bandwidth, tx power, coding scheme, and other factors. Does that mean we need calculate the data rate first?

sgalgano, 2019-10-20:
Different models may have different mechanisms for computing datarate based on waveform parameters. RF Pipe is a simple model with no such parameters. You just set the effective rate.

hideogump, 2019-10-21:
This is a question related to the RF Pipe Model.

I have a network stack consisting of the bypass MAC layer as message need to be transmitted based on time of arrival to the EMANE emulation boundary.  An external calculation will be used to determine the transmit duration based on modulation, preambles, tx modes, byte count, etc.

I am looking for an existing EMANE PHY model which provides support for tossing RF packets received over the simulated RF media whenever there is a transmit collision, to detect and ensure transmit orthogonality for multiple nodes.

The bypass PHY seems to lack any of this logic on the receive end.

Perhaps the next lightest weight PHY layer relative to the bypass would be the RF Pipe.  One area of concern is the transmit "datarate" parameter, which does not appear to support a zero value, which is to say, do not add transmit delays to tx packets.

Per the documentation, the datarate is used by the transmitter to compute the transmit delay (packet size/datarate) between successive transmissions.

Assuming I was attempting a maximum transmit rate of 100 Mbps, could I simply set the datarate to a very large number to reduce the transmit delay to some insignificant value?  Since the value is a uint64, there is a wide range of numerically high values.  

Is the delay calculated and added by the RF Pipe PHY logging to a log file for inspection of results?

hideogump, 2019-12-20:
I have observed that the transmit durations are based on a combination of the packet data size coupled with the datarate value.  I simply set the datarate value to a high enough throughput such that the RF Pipe model does not delay significantly the downstream packet, where the TDMA mechanism is based on the external entity supplying the packet.