adjacentlink/emane #124: Transmit Orthogonality Testing per existing models

hideogump, 2019-10-28:
I want to be able to test out for transmit orthogonality using one of the existing EMANE MAC or PHY models.  Notionally, if a transmit packet collides in time (meaning tx start time plus tx duration), I want the packet to be dropped, and a statistic logged.

Do all EMANE PHY models other than bypass provide this capability?

sgalgano, 2019-10-28:
There is only one physical layer model that is part of the standard emane distribution: the emulator physical layer - sometimes referred to as the framework phy. The bypass phy is an example stub with no functionality and very limited purpose.

Even though a radio model is sometimes referred to as a MAC layer, there is a portion of activity that must be performed that would notionally be considered part of physical layer processing.  It is up the the radio model to determine whether or not to discard a packet due to inband (same waveform) collision.  The framework phy provides all the necessary information but the decision is waveform (model) specific. For example SINR must be considered, collaborative transmission (constructive interference) models, etc.

hideogump, 2019-10-28:
Thanks Steve for the information.  It is safe to say that the RF Pipe Model, which is both an EMANE MAC and PHY layer implementation, provides this in band collision drop behavior, as well as out of band dropping right out of the box?  Are there any particular options required of this model to achieve this in-band tx collision dropping of packets, or does the RF Pipe Model default to this behavior being enabled?  Is this behavior entirely separate from packet dropping based on SINR and POR curves?

sgalgano, 2019-11-07:
RF Pipe is a very simple model. It does not support any in-band collision detection. Consider switching to the TDMA model which handles in-band interference.

Modeling in-band interference for non-constructive waveforms, starts with determining receiver lock  based on the earliest start-of-reception (SoR) which takes into account propagation delay. You calculate the SINR for the earliest SoR packet the same way you do normally, but in this case the overlapping in-band packet or packets count as noise signal energy and are not otherwise processed. You configure the physical layer to have [noisemode](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#noisemode) set to all, which requires you to pull the signal energy out of the reported spectrum window when calculating SINR.

closed 2019-11-07 by hideogump