adjacentlink/emane #126: Olsrd Testing in a multihop network including node mobility and frequency changes

hideogump, 2019-11-01:
I need to create a 3 node emulation session using olsrd as the routing engine, and explore both single and multihop implementations.  The nodes will be something like nodes A, B, and C.  I will be using the RF Pipe EMANE model.  

I want to start with a flat network where all nodes are connected, so A, B, and C are all on the same frequency.  I then want to migrate a node so that a relay configuration is implemented testing olsrd routing and timouts, such that A has a connection to B, and B has a connection to C, but A need to communicate one hop to C through B.  

For this to occur, I suppose I could use the pathloss event subsystem.  The changes to the network topology will be manual in human operating time through something like the EMANE shell rather than automated.

Another possibly topology would be A and B sharing the same frequency, and B and perhaps, perhaps on a different interface, sharing a frequency with C.

olsrd can target two different interfaces, but does EMANE support this single instance?

What general guidelines or recommendations would you point to such as examples for creating these test scenarios, which broadly is intended to verify olsrd network formation, timeouts, and dynamics in a multihop topology?

Would I need a custom event to change the frequency of a nodes runtime?

hideogump, 2019-11-07:
I now have this working properly using a single instance of emane with olsrd targeting this interface.  I am using the EMANE pathloss event using the RFPipe model to kill direct links between a pair of nodes by setting a large forward pathloss to check out the relay adaptive nature of olsrd, and then recovery when those links are restored.

I would also like to check out this behavior runtime by changing operating frequencies of nodes between networks?  Is that possible in EMANE runtime, i.e. frequency changes via some command line tool?

sgalgano, 2019-11-07:
You need to use a radio model that supports running-state frequency change.  There is a physical layer control message API that a radio model can use to set frequencies of interest and frequency segments for each downstream (ota) message. The RF Pipe is a simple model and does not support runtime frequency change.

You can either add the functionality you want using configuration that is running-state modifiable or you can switch to the TDMA model and use the schedule mechanism to manipulate frequency.

closed 2019-11-07 by hideogump