adjacentlink/emane #127: EMANE Shell Localhost Scope

hideogump, 2019-11-06:
While I have the EMANE shell (emanesh) command working when connecting to a specific Linux container running emane proper, I am looking as well for support at the host system level via "emanesh localhost" to allow me to get content across the simulation rather than through individual container levels.  When I attempt this connection via "emanesh localhost", this connection fails, as there is no listener.  

To get EMANE shell to operate at the localhost level, does emane proper need to be run at the host level?  There is this suggestion in the sample configuration files, but is there a particular platform.xml file needed to simply have EMANE shell support at emulation scope?

sgalgano, 2019-11-07:
`emanesh` connects to a running emulator instance's control port service which listens for incoming connections at [controlportendpoint](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#controlportendpoint).

We use [OpenTestPoint](https://github.com/adjacentlink/opentestpoint) for the functionality you [seek](https://github.com/adjacentlink/emane-tutorial/wiki/OpenTestPoint). Demonstrations contained in the [emane-tutorial](https://github.com/adjacentlink/emane-tutorial), [python-etce-tutorial](https://github.com/adjacentlink/python-etce-tutorial), and [letce2-tutorial](https://github.com/adjacentlink/letce2-tutorial) all use otestpoint to provide tiered access to statistics and tables from one or more emulation nodes (based on how you configure brokers).

You can also reference the [emane-control-port-tutorial](https://github.com/adjacentlink/emane-control-port-tutorial) to create your own tools for accessing control port information.

hideogump, 2019-11-07:
Thanks, I understand.  It would seem the use of emanesh with total emulation scope would require all NEMS to exist within a single platform.xml as in one of the examples.

closed 2019-11-07 by hideogump

sgalgano, 2019-11-07:
Keep in mind that is a highly contrived [example](https://github.com/adjacentlink/emane-tutorial/wiki/Demonstration-2) for illustrative purposes only. We do not advocate running more than one NEM in an emulator instance.