adjacentlink/emane #128: OlsrdLinkView.py Interface Requirements

hideogump, 2019-11-07:
I am interesting in using the EMANE Olsrd viewer to view dynamic links in my emulation whose routing is based on olsrd.  This emulation does not include the GPS daemon or position information.  What is required within any of the EMANE emulation radio containers running EMANE to utilize this viewer?

sgalgano, 2019-11-07:
You should be fine just running emane and olsrd with the txtinfo plugin.

hideogump, 2019-11-08:
It did not work out of the box for me.  It is not clear what EMANE options are required to run in a radio container that is mined by this application and through which interface (e.g. Telnet and http are shown in the source code).  Do the nodes need to follow a naming convention (e.g. node-001, etc.)

sgalgano, 2019-11-08:
The script expects node names like [node-NEMID](https://github.com/adjacentlink/emane-tutorial/blob/master/scripts/olsrlinkview.py#L133). It only interacts with olsrd txtinfo plugin and libgps. It is a tutorial visualization aid, you many need to modify it to meet your needs.

hideogump, 2019-11-08:
I got it to work, the name of my control interface for the radio containers was radioXctl, so I modified the source, and all is well.  I disabled the GPS thread since it is not being used.  

Is there any know way of changing the network layout such that when nodes are added, the do not lay right atop each other.  My minimum preference would be an even spatial distribution, or when GPS is enabled, the 2D layout plot might in some manner reflect the GPS coordinate?

hideogump, 2019-11-08:
I figured it out, Ctrl+S will save the node positions after the user has changed the layout.  I also added a command line argument to specify the total number of nodes in the display and to conditionally enable or disable the GPS thread.

closed 2019-11-08 by hideogump