adjacentlink/emane #129: NEM GPS Latitude and Longitude External Access

hideogump, 2019-11-09:
I need to be able to provide an external application the GPS positions associated and assigned to a NEM in an emulation container.  My understanding is that EMANE uses this information for calculating pathloss information depending upon the propagationmodel setting as either 2-ray or flatspace.   

The EMANE demos use gpsd and a pseudo-terminal to provide GPS position modelling.

What is the interface to gpsd that is used by EMANE to determine node position (e.g. Telnet)?  

Is that information available as part of the message flow between MAC and PHY layers for grabbing and forwarding to an external application?   Currently latitude and longitude are not part of the EMANE common MAC or PHY headers (to my knowledge).

sgalgano, 2019-11-10:
The emane tutorial demonstrates how you can make emulation event information available to non-emulation applications. That is what is being done with location events, emaneeventd loads one or more agent plugins, in this case [gpsdlocationagent](https://github.com/adjacentlink/emane/wiki/GPSD-Location-Agent).  

EMANE does not use libgps. 

Any NEM layer can [subscribe](https://adjacentlink.com/documentation/emane/v1.2.1/EventService.html#RegisteringForAnEvent) to any event.

You can also use the `emane.events` Python module to create event processing tools.

hideogump, 2019-11-10:
Since GPS location coordinates and node position information originates in the event subsystem, and this information is then translated into content for a running instance of gpsd, that is gpsd is driven by EMANE, it would seem that gpsd only provides the functionality of obtaining gps location information in a more easily parsed format to an external application, is this correct?  Another way of asking this question, location information is sourced within EMANE through the event subsystem, and is used by EMANE internally, and therefore gpsd might not exist as a running instance in the container itself other than providing a Telnet interface for obtaining location information to external applications, like the olsrdviewer.py application?

closed 2019-11-11 by hideogump