adjacentlink/emane #132: Obtaining Framework XML Settings in a Custom Skim

hideogump, 2019-11-11:
I need to pass some EMANE framework settings from an EMANE stack using a custom shim to an external non-EMANE application.  For example, I would like to be able to pass the frameworks systemNoiseFigure and rxSensitivity.   

This information is read at boot by the framework via an XML file and is private within the class, and does not exist as a field in the common PHY header.   

Is there a way to obtain this framework specific settings, such as systemNoiseFigure or the rxSensitivity in a custom shim?

sgalgano, 2019-11-19:
You can access configuration information externally using the [control port](https://github.com/adjacentlink/emane-control-port-tutorial/wiki/Configuration-Query), not from another NEM layer.

hideogump, 2019-12-20:
I was hoping to get this information integral to the shim stack, where this information is needed and shared rather than an external application using the control port.   I assume worst case, I could add the XML schema validation and read from the configuration file, but I wanted to determine if there was some "master" reference pointer provided to a shim, that might allow for obtaining system wide settings for a single NEM stack, that is parent emulation information.

hideogump, 2019-12-21:
Closing item

closed 2019-12-21 by hideogump