adjacentlink/emane #134: EMANE Pathloss Event for Broadcast Destination

hideogump, 2019-11-17:
The EMANE pathloss subsystem has been described as a 2 dimensional matrix between all combinations of tx NEMS and rx NEMs.  The emaneevent-pathloss Python script allows for host level setting of pathloss values runtime.  How does one set the pathloss for broadcast destination packets?   Is there any EMANE table content that provides entries for the EMANE all 1's broadcast destination?

sgalgano, 2019-11-19:
Pathloss is used when calculating [receive power](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#receive-power-calculation) at a receiver for a given source. Traffic type (unicast or broadcast) does not factor in.

hideogump, 2019-12-20:
I understand how this works on the receive side, but I need to calculate pathloss from the perspective of the transmitter so that it is available for analysis on the OTA bus from analysis of this information alone, which is transmit NEM centric rather than receive NEM centric.  I can live without it, but was thinking that if this information, that is pathloss for either multicast or broadcast destinations, that could prove to be useful to other users.

closed 2019-12-20 by hideogump