adjacentlink/emane #138: Creating Frequency Control and Timestamp Message from MAC Layer Shim

hideogump, 2019-12-19:
I have a custom MAC level shim that sits above the RF Pipe MAC/PHY element of the NEM stack.  This shim receives a combination of host data packet coupled with metadata that describes attributes associated with the packet including content that would populate the standard EMANE FrequencyControl and Timestamp messages on a per packet basis.  The NEM stack looks the following.

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE nem SYSTEM "file:///usr/share/emane/dtd/nem.dtd">
<nem>
  <transport definition="transvirtual.xml"/>
  <shim definition = "macshim.xml"/>
  <mac definition="mac.xml"/> 
  <phy> 
    <param name="frequency"                value="1300000000"/>
    <param name="frequencyofinterest"      value="1300000000"/>
```

In this custom shim, I get content from the metadata with the goal of creating a FrequencyControl and Timestamp message for passing down the stack.  

The platform does not seem to recognize these two custom populated messages by inspection of the log files.  This is true whether or not I clone the messages before passing it to the framework.

Attached is the important snippets of code from the shim's DownstreamPacket handler and the EMANE log file.

So what am I missing?

Log file:

16:50:28.397648  INFO SHIMI 001 rrpucsMac::ShimLayer::processDownstreamPacket src 1, dst 65535, size 2180, control msgs 0, verified MAC metadata: size 2088, data offset 2, MAC metadata decode 
msg id: 8738
tx sequence: 0
tx bandwidth: 9999
tx start time: 1576792228.398000
tx duration: 1000
tx center freq: 1300001
tx power: 4.000000
trailer: 8738
16:50:28.397655 DEBUG SHIMI 001 rrpucsMac::ShimLayer::processDownstreamPacket src 1, dst 65535, size 2180, adding FrequencyControlMessage msg {Freq 1300001 KHz, BW 9999 KHz, Duration 1000 us}, msgs 1
16:50:28.397660 DEBUG SHIMI 001 rrpucsMac::ShimLayer::processDownstreamPacket src 1, dst 65535, size 2180, adding TimeStampControlMessage msg {1576792228.398000}, msgs 2
16:50:28.397665 DEBUG SHIMI 001 rrpucsMac::ShimLayer::processDownstreamPacket src 1, dst 65535, size 2180, control msgs 2
16:50:28.397668  INFO SHIMI 001 rrpucsMac::ShimLayer::processDownstreamPacket: Received Non-Serialized Control Message ID=102-Frequency Control Message, FORWARDING
16:50:28.397672  INFO SHIMI 001 rrpucsMac::ShimLayer::processDownstreamPacket: Received Non-Serialized Control Message ID=110-Timestamp Control Message, FORWARDING
16:50:28.397721 DEBUG PHYI 001 FrameworkPHY::processDownstreamPacket src 1, dst 65535
16:50:28.397738 DEBUG PHYI 001 FrameworkPHY::processDownstreamPacket Frequency Control Message bandwidth: 0 freq: 0 duration: 87 offset: 0 power: 0.000000 
16:50:28.397743 DEBUG PHYI 001 FrameworkPHY::processDownstreamPacket Time Stamp Control Message timestamp 1576792228.397691
16:50:28.397756 DEBUG PHYI 001 FrameworkPHY::processDownstreamPacket Common PHY Header regid: 7 seq: 0 bandwidth: 10000000 fixed antenna gain: on fixed antenna gain: 0.000000 tx time: 1576792228.397691 freq: 1300000000 duration: 87 offset: 0 src: 1 transmitter power: 0.000000 
16:50:28.397804  INFO SHIMI 001 eViewPhy::ShimLayer::processDownstreamPacket Common PHY Header regid: 7 seq: 0 bandwidth: 10000000 fixed antenna gain: on fixed antenna gain: 0.000000 tx time: 1576792228.397691 freq: 1300000000 duration: 87 offset: 0 src: 1 transmitter power: 0.000000  size 2249

Source Code snippet (only FrequencyControlMessage shown, TimeStamp is similar):
```

void EMANE::Models::rrpucsMac::ShimLayer::processDownstreamPacket(DownstreamPacket &pkt, const ControlMessages &msgs)
{
    try
    {
        // Create a non-constant collection of messages from passed msgs
        ControlMessages ctlMsgs = msgs;

                    // Create a frequency control message from fields in the MAC Layer metadata
                    if (pMacMetadata->m_u32FrequencyKHz && pMacMetadata->m_txDurationUs)
                    {
                        uint64_t u64BandwidthHz = (uint64_t)pMacMetadata->m_u16BandwidthKHz * 1000;
                        uint64_t u64FrequencyHz = (uint64_t)pMacMetadata->m_u32FrequencyKHz * 1000;
                        double dPowerdBm = pMacMetadata->m_txPowerdBm;

                        timeval tvDuration = makeTimeval(0, pMacMetadata->m_txDurationUs);
                        tvDuration = normalizedTimeval(tvDuration);
                        Microseconds durationMicroseconds = timevalToDuration(tvDuration);

                        // Create single frequency segment at specified freq, tx power, and tx duration
                        FrequencySegment segment(u64FrequencyHz,
                                                 dPowerdBm,
                                                 durationMicroseconds);

                        FrequencySegments frequencySegments;
                        frequencySegments.push_back(segment);

                        // Creates a FrequencyControlMessage on the heap
                        EMANE::Controls::FrequencyControlMessage *fcm = EMANE::Controls::FrequencyControlMessage::create(u64BandwidthHz, frequencySegments);

                        // Add control message to list
                        ctlMsgs.push_back(fcm);
                    }

        // Log control messages
        logControlMessages(ctlMsgs, __func__);

        // Send pkt to lower layer with control messages
        sendDownstreamPacket(pkt, clone(ctlMsgs));

```

sgalgano, 2019-12-20:
Control messages are exchanged between contiguous layers.

hideogump, 2019-12-20:
So how then do I set the framework with run time attributes associated with both the frequency control and timestamp messages?  Can it be done or is it simply not supporting using the shim based architecture?

sgalgano, 2019-12-20:
You can only send a control message to the physical layer from the layer above it in the stack. In this case, based on your nem XML, that would be the MAC (radio model) plugin. My suggestion would be to investigate writing your own radio model, absorbing in whatever you are doing in your shim.

hideogump, 2019-12-20:
Thanks Steve, for the quick response.

Just for clarity, my mac.xml is as follows which is the standard RF Pipe implementation.

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE mac SYSTEM "file:///usr/share/emane/dtd/mac.dtd">
<mac library="rfpipemaclayer">
  <param name="enablepromiscuousmode" value="off"/>
  <param name="datarate"              value="200M"/>
  <param name="jitter"                value="0"/>
  <param name="delay"                 value="0"/>
  <param name="flowcontrolenable"     value="off"/>
  <param name="flowcontroltokens"     value="10"/>
  <param name="pcrcurveuri"           value="file:///usr/share/emane/xml/models/mac/rfpipe/rfpipepcr.xml"/>
</mac>

```
Could I achieve what I want to do by moving the custom MAC shim between the MAC and PHY layers of the NEM stack similar to the following to get in between the MAC and Framework's PHY?

```
<nem>
  <transport definition="transvirtual.xml"/>
  <mac definition="mac.xml"/> 
  <shim definition = "rrpucsmacshim.xml"/>
```

sgalgano, 2019-12-20:
Yes. If your shim sits above the physical layer, it can send physical layer control messages downstream.

hideogump, 2019-12-20:
Thanks, seemed to work as I intended with these changes.  I now view my metadata passed content in the FrameworkPHY logging.

Thanks as always for the help, closing this one out.

closed 2019-12-20 by hideogump