adjacentlink/emane #139: Transmit Power vs Segment Power for RF Pipe Model RF Sensitivity

hideogump, 2019-12-20:
The RF Pipe model includes a _txpower_ field in XML, presumably read-only, whereas each transmission composed of one or more segments contains a segment power.  Reception is based on rx sensitivity relative to _rxPower_, where rx power includes in the spec _txPower._

So the question then is, what is the relationship for rx reception between these two settings, one somewhat static, and the other per packet (or combined from multiple segments) based?

From inspection of the RF Model source code, it appears that reception is based on the segment specific power, rather than tx power.  If that is the case, is the txpower setting meaningless?

Both are presumably carried as information on the Common PHY Header or OTA frame.

sgalgano, 2019-12-20:
[txpower](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#txpower) is a physical layer configuration parameter.

When a radio model sends an over-the-air packet downstream it uses a [FrequencyControlMessage](https://github.com/adjacentlink/emane/blob/master/include/emane/controls/frequencycontrolmessage.h) to specify duration using one or more [frequency segments](https://github.com/adjacentlink/emane/blob/master/include/emane/frequencysegment.h). If a radio model wishes, it can set a per segment tx power in dBm. A radio model can also send a [TransmitterControlMessage](https://github.com/adjacentlink/emane/blob/master/include/emane/controls/transmittercontrolmessage.h) (constructive-interfrence/collaborative-transmission) along with the downstream packet, to specify a power level for each transmitter. It neither control message is specified, the physical layer will use [txpower](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#txpower) as the power level for each segment.

The order of precedence (highest to lowest):

1.  per segment power
2.  per transmitter power
3.  [txpower](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#txpower).

RF Pipe is a pretty simple model. More complex models have more complex needs.

hideogump, 2019-12-20:
Thanks for the clarification

hideogump, 2019-12-21:
This item is closed

closed 2019-12-21 by hideogump