adjacentlink/emane #140: EMANE Stack Control Message Clone Rules

hideogump, 2019-12-20:
I am composing several custom skims that make part of the EMANE NEM stack, and they are arranged at various locations in the stack, above the MAC, between MAC and PHY, and below the PHY.

Each of these shims has handlers for the processUp/DownstreamPacket and Control.  

For most of these handlers, the skim does nothing with the control messages and preferable acts as a conduit between adjacent layers.

Under a rare occasion, some layer specific control messages are composed on the heap and substituted for a nominal EMANE control message (e.g. FrequencyControl or Timestamp).

I have by practice cloned the message similar to the following code snippet.  

I am not sure I fully understand the rules for when control messages need to be cloned in the shim, and when not to.  There are occasions where operation seems to require the clones, and other cases where this does not seem to be a problem.

In all cases, the control messages passed through the stack are not freed by the custom skim, at best inspected, or perhaps substituted for.

So what are the hard rules for when to clone and when not to?  Does this logic apply equally to control messages that pass through the Packet and Control methods?

I want to avoid any form of memory leaks when operating the emulation for extended durations with heavy packet loads.

    EMANE::ControlMessages clones;
    for (const auto &msg : msgs)
    {
        clones.push_back(msg->clone());
    }
    return clones;

sgalgano, 2019-12-20:
If you wish to store a control message that you received (`process{Up,Down}stream{Packet,Control}`) or prior to sending  (`send{Up,Down}stream{Packet,Control}`) you must clone the message. Ownership of a control message as well as deallocation responsibility transfer to the emulator framework.

You are responsible for cleaning up all created or cloned control messages that have not been passed to the emulator framework for upstream or downstream delivery  (`send{Up,Down}stream{Packet,Control}`) .

hideogump, 2019-12-20:
Just to make sure I do not misunderstand, since I am not storing any of the control messages, if I simply pass the control messages either downstream or upstream, they must be cloned as part of the desire to simply pass the messages through the layers?  So notionally, is the idea that the framework gives the message to say a custom shim, and it thereafter deletes that message assuming it was consumed?  If I want to reduce the amount of possible cloning, since I am not currently adding any additional logic to the sendUp/DownControl methods, would it be simpler to simply not have these handlers in place in the shims, or are they necessary to hook up the messaging between layers?

sgalgano, 2019-12-20:
If you are forwarding a copy of a control message between layers, you must clone the received message before sending it with `send{Up,Down}stream{Packet,Control}`.

hideogump, 2019-12-21:
This item is closed

closed 2019-12-21 by hideogump