adjacentlink/emane #141: freespace model not working

kammirzazad, 2019-12-25:
I am trying to have wireless nodes at varying distances from a master node through location events to emulate different link characteristics. However, independent of the distance, I dont see any difference in ping times nor loss rates nor throughputs. I am wondering if there is anything I am missing? 
I have already enabled event agent and gpsd in each nem.

hideogump, 2019-12-31:
I can only comment on my experience.  Using the RF Pipe EMANE model, and the default PCR curve (which is linear), when I position modes with freespace calculated pathloss values that correspond to a certain percentage of loss rates, I am able to get packet drops that over time average out to be what is specified in the PCR curve.  Steve and company can comment, but from looking through the EMANE source code, the drop rate has a random nature to it, so instantaneous drop rates might not be exactly the number associated with the pathloss based on the freespace distance between two nodes.  You do have to start out with knowing what your receive sensitivity is and your segment receive power.  In my case, the PCR curve ranges from a zero loss pathloss of 80 dB to 100 percent loss at 100 dB pathloss.

kammirzazad, 2020-01-17:
@hideogump Thank you for your comment. After trying couple of different scenarios, I realized that to see any loss due to pathloss, one either needs a very long distance or higher (unicast) rate.  I was able to observe losses with 54Mbps unicast rate at ~150 meters distance. When nakagami fading is enabled, losses happen even in shorter distances (~30 meters). 
Now that I have succeeded to create different loss rates, I realize EMANE does not support ACK frames nor retransmissions (#145). How did you handle losses?

closed 2020-01-17 by kammirzazad