adjacentlink/emane #142: EMANE Reception Logic when tuned slighly off center frequency

hideogump, 2019-12-31:
I am trying to simulate a 10 MHz OFDM waveform, which will have a small range of tuning off frequency where it should be received by the receiver, that is a frequency deviation.  In the EMANE model, it is being modeled as a single frequency segment with a particular center frequency, say 1.3 GHz, and a 10 MHz bandwidth.

How does EMANE handle signal reception when the receiver _frequencyofinterest_ has some slight center frequency deviation relative to the tx signal's tx frequency?

Is there a way to tune or customized this drop behavior for in-band signals to match my waveform?

sgalgano, 2020-01-02:
Your best option is to add functionality to your radio model that instructs receivers to apply some offset and then as a result of that offset, take some action such as decreasing PCR.  For offset applied at the receiver, you can use a random amount, a fixed amount for all receivers, a fixed amount but different for each receiver, etc.

hideogump, 2020-01-02:
Thanks for the explanation

closed 2020-01-02 by hideogump