adjacentlink/emane #144: How Over-the-air(OTA) multicast channel works

ChengDaHaI, 2020-01-13:
I am confused about how OTA manager and OTA multicast channel works. 
From [wiki](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#otamanagerdevice), otamanagerdevice seems to be the network interface who transmits and receives OTA messages (also accounts for the OTA channel). From the wiki definition of [otamanagergroup](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#otamanagergroup), it is like a IP address and port number pair which used for (socket) OTA communication. However, in the example of emane tutorial,  otamanagergroup is 224.1.2.8:45702, which is also given in the figure in Wiki introduction. 

So, could you elaborate how otamanagergroup(224.1.2.8:45702) is related with otamanagerdevice (for example, eth0 in tutorial: 10.99.0.1) and how OTA channel works?

P.S. a typo in wiki: 
```
otamanagergroup
IPv4 or IPv6 Event Service OTA channel endpoint.
```
It should not be _Event Service_.

sgalgano, 2020-01-14:
The [otamanagerdevice](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#otamanagerdevice) specifies the interface you want your [otamanagergroup](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#otamanagergroup) (multicast) traffic routed over. Without specifying an interface, the kernel will consult the route table for a mutlicast route or if not found, the default route.

The tutorial is configuring both the  OTA and event multicast devices to be the lxc container node's *eth0* which is part of the *emanenode0* host bridge.

ChengDaHaI, 2020-01-14:
> The [otamanagerdevice](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#otamanagerdevice) specifies the interface you want your [...]

Thank you for the replay.

Can i rephrase your reply as  otamanagerdevice specifies the interface where the OTA message (the traffic from emulated interfaces, like emane0 in tutorial) routed over ? If so, the OTA multicast channel  is supported (actually i don't know use what verb) by the _emanenode0_ host bridge network, right? 

The question you don't mention is why otamanagergroup is defined in the form of _224.1.2.8:45702_. Is there any reason why it is defined in this form? It looks wierd  since no network interface matching the ip subnet _224.1.2.8_.

sgalgano, 2020-01-14:
I think it would be beneficial to gain some background on [IP multicast](https://en.wikipedia.org/wiki/IP_multicast).

ChengDaHaI, 2020-01-14:
Thank you very much!

closed 2020-01-23 by ChengDaHaI