adjacentlink/emane #145: 802.11 ACK frames

kammirzazad, 2020-01-17:
Looking at the source code for 802.11 MAC layer, I dont see any support for emulating ACK frames or transmissions. I was wondering if it is implemented anywhere else. If not, is there any plan to add support for these frames in future?

patelkb, 2020-01-17:
The EMANE 80211abg model does not actually implement the CSMA-CA
protocol due to timing restrictions within a real-time emulation.

The model emulates unicast transmissions (RTS-CTS-MSG-ACK or MSG-ACK) as
well as retries without transmitting the control messages or the failed
data message. The time on air for a given transmission does take into
account the entire control and data exchange.  Unicast with multiple
retries is handled on the receive side by rolling the dice multiple
times (as required) with the received probability of reception.

The model emulates collisions based on estimation of channel activity
from both one and two hop neighbors.

There are no plans to update the 80211abg model.

Kaushik B. Patel
Adjacent Link LLC

kammirzazad, 2020-01-21:
Thank you @patelkb.

closed 2020-01-21 by kammirzazad