adjacentlink/emane #146: Frequency Hopping/Agility in EMANE

hideogump, 2020-01-18:
Is there any support in EMANE through any existing model for implementing a frequency hopping based system?  Presumably, nodes would need to have a hop set defined, be synchronized in time, and have the hop set frequency tied to some slot time.

If there is no direct support for frequency agility, to receive signals on various frequencies, is it a requirement to define the range of frequencies and the EMANE PHY _frequencyOfInterest_ parameters where the hopset is know apriori?  For this implementation, presumably packets received would not be dropped because the receive is on the wrong frequency of the hopset.

sgalgano, 2020-01-18:
When we model frequency hopping waveforms, we set the [frequencyofinterest]( https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#frequencyofinterest) to include all possible hopping frequencies. The radio model implementation uses all or a subset of the [frequencyofinterest]( https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#frequencyofinterest)  set when building over-the-air message frequency segments (hopping pattern).

hideogump, 2020-01-18:
Thanks, much as I expected.  So notionally, the receiver then would be preconfigured for all possible frequencies, but then there is no sense of the receiver being in lock step with the transmitters hop frequency set?  Assuming you wanted that type of behavior, I assume that might require a custom model and or a custom shim?

sgalgano, 2020-01-18:
You would need logic in your radio model that enforces the use of the correct pattern over time. The radio model would be responsible for dropping frequency segments that are invalid for a given pattern.

hideogump, 2020-01-19:
I understand, thanks.

closed 2020-01-19 by hideogump