adjacentlink/emane #153: EMANE Top Hat Application

hideogump, 2020-02-19:
I have perhaps a somewhat unique need.  I want to create a standalone application that runs as a daemon within a radio container along with EMANE.  This application, needs to behave like a top hat to EMANE, creating a TAP interface similar to EMANE's _emane0_, say _tophat0_.  

The purpose of this architecture is to allow an external application to provide transmission buffers for downstream packets and QoS behavior external to EMANE, and have it deliver data according to some unique and dynamic requirements, which may include TDMA, carrier detect, etc.

I do not want to incorporate the significant levels of development for this external component into something like an EMANE shim or custom model if possible.

This TAP interface would need to be targeted by the Linux kernel routing rather than _emane0_.  The _olsrd_ configuration file would target this interface as tophat0 rather than _emane0_, so that the kernel would see this virtual Ethernet interface and the radio's interface.  The _emane0_ interface would then be controlled directly by the top hat application which creates tophat0, and would directly read and write to the emane0 TAP interface.  

I do not want the kernel to write directly to the _emane0_ interface, and I do not want EMANE to send packets to the kernel, but instead pathway to the kernel would be through the top hat application.  

Creating the TAP interface _tophat0_ is easy, and I am in the process of getting the file descriptor for the running or attached instance of _emane0_.  Using this file descriptor, I was hoping to read and write directly to the _emane0_ TAP interface.

Does this approach seem like something that would work?  Is there anything special that needs to be done to say the emane0 interface to achieve this behavior (e.g. mark it as down to the OS so that it can be written to directly only by the tophat application)?

What are your recommendations?

sgalgano, 2020-02-19:
You can create a custom transport using the emane API, you can [embed](https://adjacentlink.com/documentation/emane/v1.2.1/index.html#FAQ) a transport in your application, or you can set the emulator to external transport and send wire-level messages directly to the emulator (hitting the NEM stack at the radio model) using UDP or TCP. Using wire-level messages will involve reviewing the code to determine the formats, it is not part of the documented API.

hideogump, 2020-02-19:
Thanks Steve as always for the prompt response.  My desire would not to have to link in EMANE library components and routine into the top hat application if possible, but that is not a show stopper.  If I could write it as a component that could be used, that would be different. 

In my TAP bridge testing (bridging custom TAP interface to EMANE TAP interface), I was unable to "attach" to the existing EMANE TAP interface properly without having read errors even though it was supposed to use similar APIs to tunnel creation (https://backreference.org/2010/03/26/tuntap-interface-tutorial/).  

Can you point me in the general direction for information on writing a custom transport or perhaps some example as I have not traversed down that particular road (yet).

hideogump, 2020-02-27:
I was able to implement this top hat logic with a custom TAP interface subordinating the emane0 TAP interface.   It reqiured a raw socket for the emane0 interface and some IP table rules to achieve this solution which did not require use of the emane libraries.  One could argue that this could also readily be implemented as suggested as a custom transport, which might be the more elegant solution.  Here I was bound to not using the emane libraries as the solution as a guideline, but that requirement might need to be re-evaluated.

closed 2020-02-27 by hideogump