adjacentlink/emane #156: Specifying Packet Delay Characteristics for a Specific Link Runtime Dynamically

hideogump, 2020-03-19:
I would like to set the packet delay characteristics for a particular link runtime, something similar to what can be performed by driving the emulation through pathloss or location events.  While there are packet delays characteristics for models such as the RF Pipe, those settings would apply to all packets, and I want this behavior to be link specific, but dynamically controlled through some form of emulation driving script.  The CommEffect shim has some of this behavior, but I need for it to be link specific.

What do you guys recommend?

sgalgano, 2020-03-24:
The CommEffect model, while not a radio model, will allow you to set per link delays and modify them while running. If you are looking for a similar feature in a radio model you are either going to have to modify one of the existing models or roll your own.

hideogump, 2020-03-25:
Thanks, will look into it

closed 2020-03-25 by hideogump