adjacentlink/emane #166: Error During Building From Source

ChengDaHaI, 2020-04-24:
Due to specific requirements, I need install Emane 0.9.3 to test my code. 

- I download the source code from https://github.com/adjacentlink/emane/archive/v0.9.3.zip. 
- Install the dependencies for Ubuntu 16.04 
-  "./autogen.sh && ./configure && make ". 

The error occurs in "./configure":
`checking ace/OS_main.h usability... no
checking ace/OS_main.h presence... no
checking for ace/OS_main.h... no
configure: error: "Missing ACE development headers"
`
After trying `make`, i get "make: *** No targets specified and no makefile found.  Stop.". I also try `make deb` but it also doesn't work.

Thanks

b00ga, 2020-04-24:
EMANE 0.x requires the ACE library. Ubuntu 16.04 includes ACE as part of the standard Ubuntu repos.
```apt install libace-dev```

ChengDaHaI, 2020-04-28:
> EMANE 0.x requires the ACE library. Ubuntu 16.04 includes ACE as part of the standard Ubuntu repos.
> `apt install libace-dev`

Thank you for the reply. 

In the last step of building "make deb",  it raises error in building emanesh deb:

`
make[3]: Entering directory '/home/sc-ai/emane-0.9.3/.debbuild/emane-0.9.3/src/emanesh/deb_dist/emanesh-0.9.3'
python setup.py install --force --root=debian/emanesh --no-compile -O0 --install-layout=deb --prefix=/usr
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --install-layout not recognized
debian/rules:22: recipe for target 'override_dh_auto_install' failed
make[3]: *** [override_dh_auto_install] Error 1
make[3]: Leaving directory '/home/sc-ai/emane-0.9.3/.debbuild/emane-0.9.3/src/emanesh/deb_dist/emanesh-0.9.3'
debian/rules:7: recipe for target 'binary' failed
make[2]: *** [binary] Error 2
make[2]: Leaving directory '/home/sc-ai/emane-0.9.3/.debbuild/emane-0.9.3/src/emanesh/deb_dist/emanesh-0.9.3'
dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2
`

Do you have any idea abou the error? 

My python version is 2.7.18 and OS is Ubuntu 16.04.

Thanks.

legit13, 2020-05-05:
Thanks @b00ga , I had the same issue and your command also fixed that issue for me.

Afterwards I got a similar error as @ChengDaHaI 

`LINE:  -- Adjacent Link LLC <emane@adjacentlink.com>  Thu, 05 Aug 2010 08:47:45 -0400
dpkg-buildpackage: full upload; Debian-native package (full source is included)
make -C .debbuild/emane-0.9.3/src/emanesh deb && \
         cp .debbuild/emane-0.9.3/src/emanesh/deb_dist/emanesh_0.9.3-1_all.deb .debbuild
make[1]: Entering directory '/home/yourname/myrepo_emanecore/EMANE/emane-0.9.3/.debbuild/emane-0.9.3/src/emanesh'
if test -f stdeb.cfg; then chmod u+w stdeb.cfg; fi
sed -e 's|@VERSION[@]|0.9.3|g' -e 's|@PYTHON[@]|python|g' -e 's|@DEBIAN_VERSION[@]|stretch|g' stdeb.cfg.in > stdeb.cfg
chmod g-w,u-w stdeb.cfg
python setup.py --command-packages=stdeb.command bdist_deb
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help`
`
error: invalid command 'bdist_deb'
Makefile:607: recipe for target 'deb' failed
make[1]: *** [deb] Error 1
make[1]: Leaving directory '/home/yourname/myrepo_emanecore/EMANE/emane-0.9.3/.debbuild/emane-0.9.3/src/emanesh'
Makefile:894: recipe for target 'deb' failed
make: *** [deb] Error 2
`

b00ga, 2020-05-06:
You need to install python-stdeb.

If you've already tried to build, I found I had to start over to make sure everything gets properly detected:
```
make distclean
./configure
make deb
```