adjacentlink/emane #171: How to setup the distance in the IEEE80211abg model?

robertclf, 2020-07-04:
It's a question, please:

I have an IEEE80211abg model with two nodes that are apart from each other X meters, using GPS coordinates with LocationEvent. Then I execute a "ping radio-2" from node-1 and shows that it can reach the node-2. How do I setup this model so that if I move the node-2 to X+100 meters away, the "ping radio-2" will return only "Destination Host Unreachable"?

My current configuration is:

```
<mac library="ieee80211abgmaclayer">
  <param name="mode" value="1"/>
  <param name="enablepromiscuousmode" value="off"/>
  <param name="distance" value="1000"/>
  <param name="unicastrate" value="6"/>
  <param name="multicastrate" value="6"/>
  <param name="rtsthreshold" value="0"/>
  <param name="pcrcurveuri" value="ieee80211abg_pcr.xml"/>
  <param name="flowcontrolenable" value="off"/>
  <param name="flowcontroltokens" value="10"/>
</mac>

<phy>
    <param name="bandwidth" value="20M"/>
    <param name="frequency" value="2.4G"/>
    <param name="frequencyofinterest" value="2.4G"/>
    <param name="subid" value="1"/>
    <param name="systemnoisefigure" value="4.0"/>
    <param name="txpower" value="0.0"/>
    <param name="noisemode" value="outofband"/>
    <param name="noisebinsize" value="20"/>
    <param name="propagationmodel" value="precomputed"/>
</phy>
```

I can see that node-2 is unreachable when I move it away very very big distances, but I need this to happen when moving away ~100 meters.

Thanks in advance!

WorksWellWithOthers, 2020-07-04:
Have you tried lowering your tx power?

robertclf, 2020-07-04:
I tried it now, with different txpower values but it shows me similar packet loss percentages when the nodes are in the same location and when they are more than 800 meters apart. Also, I tried changing the distance parameter to 500 and it returns similar packet loss percentages, too.

WorksWellWithOthers, 2020-07-04:
How about when you adjust antenna pattern gains? Does that help?

patelkb, 2020-07-04:
The below 2 equations determine your SINR which are used to determine
your packet completion rate based on the multicast/unicast rate set to 6
in your configuration.

rxPower = txPower + txAntennaGain + rxAntennaGain − pathloss
rxSensitivity = −174 + noiseFigure + 10log(bandWidth)

    <datarate index="6">
      <row sinr="-1.0"  por="0.0"/>
      <row sinr="0.0"   por="0.3"/>
      <row sinr="1.0"   por="10.5"/>
      <row sinr="2.0"   por="50.3"/>
      <row sinr="3.0"   por="84.9"/>
      <row sinr="4.0"   por="97.5"/>
      <row sinr="5.0"   por="99.7"/>
      <row sinr="6.0"   por="100.0"/>
    </datarate>

Based on your configuration (assuming antennagains are set to 0):
   rxPower = -pathloss
   rxSensitivity = -96.99

Which means your SINR = -pathloss + 96.99.

Your "propagationmodel" setting is set to "precomputed" so you are
sending that in via pathloss events.  If that is the intention, send in
a pathloss of 90 when the two nodes are a distance of X meters apart and
a value of 96.9 when the two nodes are a distance of (X+100) meters apart.

Another option is to change "propagationmodel" configuration to
"freespace" or "2ray" and compute the pathloss for those models and make
sure your SINR values come out the way you want.  You can set txpower
and antennagains to achieve desired results here as well.

Note: The "distance" parameter in the radio model configuration is for
slot budgeting and has no impact on pathloss and receive power.

kb

Kaushik B. Patel
Adjacent Link LLC

robertclf, 2020-07-05:
Thank you very much for that explanation!

closed 2020-07-05 by robertclf