adjacentlink/emane #172: IDLE slot in TDMA frame

WizKern, 2020-07-15:
The point 4 of the documentation [https://github.com/adjacentlink/emane/wiki/TDMA-Model#Defining-a-TDMA-Schedule](url) “For a full schedule, any slot not defined in a frame will be a receive slot for all NEMs referenced in the schedule. … “ is erroneous.
It should read:
“For a full schedule, any slot not defined in a frame will be a receive IDLE slot for all NEMs referenced in the schedule. …”
 
I discovered this issue about IDLE slots using a TDMA schedule with the last EMANE version 1.2.6 as well as for the earlier version 1.0.1.
Any node not defined in a TDMA frame will be set as an IDLE node.
Example and workaround: if the desired schedule is Tx1Tx2-Tx1Tx1 (frame1-frame2) to send 3 slots to node 2, it should be written Tx1Tx2-Tx1Rx2Tx1 otherwise node 2 will receive data only once instead of the expected 3 times.
Rx2 can be set using any index in the data frame; index 0 in the example given below.

frame2:

   `<frame index='1' datarate='11M'>`
      `<slot index='0' nodes='1'>`
        `<tx/>`
      `</slot>`
      `<slot index='0' nodes='2'>`
        **`<rx/>`**
      `</slot>`
      `<slot index='1' nodes='1'>`
        `<tx/>`
    `</frame>`

Yann & Fred

sgalgano, 2020-07-28:
The current documentation correctly captures the way TDMA XML schedules work.
 
The following schedule XML:

```xml
<emane-tdma-schedule >
  <structure frames="1"
             slots="5"
             slotoverhead="0"
             slotduration="1500"
             bandwidth="1M"/>
  <multiframe frequency="2400000000"
              power="0"
              class="3"
              datarate="11M">
   <frame index="0"  frequency="2420000000">
     <slot index="0" nodes="1"/>
     <slot index="1" nodes="2"/>
     <slot index="2" nodes="3"/>
     <slot index="3" nodes="4"/>
     <slot index="4" nodes="5"/>
   </frame>
  </multiframe>
</emane-tdma-schedule>
```

Results in the following at each NEM:

```
$ for i in $(seq 1 5); do  emanesh node-$i get table nems mac scheduler.ScheduleInfoTable; done
nem 1   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency  | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | TX   | 2420000000 | 11000000  | 0.0   | 3     | 0           |
| 1     | 0     | 1    | RX   | 2420000000 |           |       |       |             |
| 2     | 0     | 2    | RX   | 2420000000 |           |       |       |             |
| 3     | 0     | 3    | RX   | 2420000000 |           |       |       |             |
| 4     | 0     | 4    | RX   | 2420000000 |           |       |       |             |

nem 2   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency  | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | RX   | 2420000000 |           |       |       |             |
| 1     | 0     | 1    | TX   | 2420000000 | 11000000  | 0.0   | 3     | 0           |
| 2     | 0     | 2    | RX   | 2420000000 |           |       |       |             |
| 3     | 0     | 3    | RX   | 2420000000 |           |       |       |             |
| 4     | 0     | 4    | RX   | 2420000000 |           |       |       |             |

nem 3   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency  | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | RX   | 2420000000 |           |       |       |             |
| 1     | 0     | 1    | RX   | 2420000000 |           |       |       |             |
| 2     | 0     | 2    | TX   | 2420000000 | 11000000  | 0.0   | 3     | 0           |
| 3     | 0     | 3    | RX   | 2420000000 |           |       |       |             |
| 4     | 0     | 4    | RX   | 2420000000 |           |       |       |             |

unable to connect to node-4:47000
nem 5   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency  | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | RX   | 2420000000 |           |       |       |             |
| 1     | 0     | 1    | RX   | 2420000000 |           |       |       |             |
| 2     | 0     | 2    | RX   | 2420000000 |           |       |       |             |
| 3     | 0     | 3    | RX   | 2420000000 |           |       |       |             |
| 4     | 0     | 4    | TX   | 2420000000 | 11000000  | 0.0   | 3     | 0           |
```

I believe you are confusing ([here](https://github.com/adjacentlink/emane/wiki/TDMA-Model#defining-a-tdma-schedule)):

3. For a full schedule, any frame not defined will be idle for all NEMs referenced in the schedule.

4. For a full schedule, any slot not defined in a frame will be a receive slot for all NEMs referenced in the schedule.

If the schedule had more than 1 frame per multiframe (`frames="2"`):

```xml
<emane-tdma-schedule >
  <structure frames="2"
             slots="5"
             slotoverhead="0"
             slotduration="1500"
             bandwidth="1M"/>
  <multiframe frequency="2400000000"
              power="0"
              class="3"
              datarate="11M">
   <frame index="0" frequency="2420000000">
     <slot index="0" nodes="1"/>
     <slot index="1" nodes="2"/>
     <slot index="2" nodes="3"/>
     <slot index="3" nodes="4"/>
     <slot index="4" nodes="5"/>
   </frame>
  </multiframe>
</emane-tdma-schedule>
```

And you do not define frame `index="1"` (0-based index), you will see idle slots:

```
$ emanesh node-1 get table nems mac scheduler.ScheduleInfoTable
nem 1   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency  | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | TX   | 2420000000 | 11000000  | 0.0   | 3     | 0           |
| 1     | 0     | 1    | RX   | 2420000000 |           |       |       |             |
| 2     | 0     | 2    | RX   | 2420000000 |           |       |       |             |
| 3     | 0     | 3    | RX   | 2420000000 |           |       |       |             |
| 4     | 0     | 4    | RX   | 2420000000 |           |       |       |             |
| 5     | 1     | 0    | IDLE |            |           |       |       |             |
| 6     | 1     | 1    | IDLE |            |           |       |       |             |
| 7     | 1     | 2    | IDLE |            |           |       |       |             |
| 8     | 1     | 3    | IDLE |            |           |       |       |             |
| 9     | 1     | 4    | IDLE |            |           |       |       |             |
```

closed 2020-07-28 by sgalgano

WizKern, 2020-07-31:
Your example works fine because 2 frames are declared in the `<structure>` but only the first frame is defined.

The confusion comes when we declare 2 frames and define 2 frames. Therefore we still believe, it would be helpful to clarify the documentation. Please find below the detailed example demonstrating the unexpected IDLE slot behaviour.

The following schedule as an example to show (unexpected to us) IDLE slots for node 2 in the second frame.
The full schedule is defined to use 2 frames and both of them are specified in the XML file as follow.
```xml
<emane-tdma-schedule>
  <structure bandwidth="250K" frames="2" slotduration="5800" slotoverhead="3" slots="2" />
  <multiframe datarate="1M" frequency="300M" power="47.0">
    <frame index="0">
      <slot index="0" nodes="1">
        <tx class="3" />
      </slot>
      <slot index="1" nodes="2">
        <tx class="3" />
      </slot>
    </frame>
    <frame index="1">
      <slot index="0" nodes="1">
        <tx class="0" />
      </slot>
      <slot index="1" nodes="1">
        <tx class="0" />
      </slot>
    </frame>
  </multiframe>
</emane-tdma-schedule>
```

As we can see in the second frame the node 2 is set IDLE.
Considering the documentation we should expect the node 2 to be RX.
I quote from the documentation : "4. For a full schedule, any slot not defined in a frame will be a receive slot for all NEMs referenced in the schedule."
```
nem 1   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | TX   | 300000000 | 1000000   | 47.0  | 3     | 0           |
| 1     | 0     | 1    | RX   | 300000000 |           |       |       |             |
| 2     | 1     | 0    | TX   | 300000000 | 1000000   | 47.0  | 0     | 0           |
| 3     | 1     | 1    | TX   | 300000000 | 1000000   | 47.0  | 0     | 0           |

nem 2   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | RX   | 300000000 |           |       |       |             |
| 1     | 0     | 1    | TX   | 300000000 | 1000000   | 47.0  | 3     | 0           |
| 2     | 1     | 0    | IDLE |           |           |       |       |             |
| 3     | 1     | 1    | IDLE |           |           |       |       |             |
```
We can observe: at indexes 2 and 3 the type for nem 2 is IDLE and not Receive as expected.
In order to solve this issue, any slot that is NOT defined in the frame has to be set as a receive slot.
The following full schedule shows the workaround with the results below.
```xml
<emane-tdma-schedule>
  <structure bandwidth="250K" frames="2" slotduration="5800" slotoverhead="3" slots="2" />
  <multiframe datarate="1M" frequency="300M" power="47.0">
    <frame index="0">
      <slot index="0" nodes="1">
        <tx class="3" />
      </slot>
      <slot index="1" nodes="2">
        <tx class="3" />
      </slot>
    </frame>
    <frame index="1">
      <slot index="0" nodes="1">
        <tx class="0" />
      </slot>
      <slot index="0" nodes="2">
        <rx />
      </slot>
	  <slot index="1" nodes="1">
        <tx class="0" />
      </slot>
    </frame>
  </multiframe>
</emane-tdma-schedule>
```
```
nem 1   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | TX   | 300000000 | 1000000   | 47.0  | 3     | 0           |
| 1     | 0     | 1    | RX   | 300000000 |           |       |       |             |
| 2     | 1     | 0    | TX   | 300000000 | 1000000   | 47.0  | 0     | 0           |
| 3     | 1     | 1    | TX   | 300000000 | 1000000   | 47.0  | 0     | 0           |

nem 2   mac scheduler.ScheduleInfoTable
| Index | Frame | Slot | Type | Frequency | Data Rate | Power | Class | Destination |
| 0     | 0     | 0    | RX   | 300000000 |           |       |       |             |
| 1     | 0     | 1    | TX   | 300000000 | 1000000   | 47.0  | 3     | 0           |
| 2     | 1     | 0    | RX   | 300000000 |           |       |       |             |
| 3     | 1     | 1    | RX   | 300000000 |           |       |       |             |
```
Please let us know if we misunderstood something.

Best regards,

Yann & Fred.