adjacentlink/emane #175: Utilizing EMANE in constructive testbed

Nocsaron, 2020-09-04:
I am interesting in using EMANE to better model wireless communications in a constructive simulation testbed.  Is it possible to "drive" the positions & send/receive messages via an external simulation?

patelkb, 2020-09-04:
Yes, your simulator can drive the emulator via the defined APIs.

1.   Pathloss within the physical layer model is based on location or
pathloss events. Pathloss is dynamically calculated based on location
events when the propagationmodel configuration parameter is set to
either 2ray or freespace , which selects between the 2-ray flat earth or
freespace propagation models, respectively.
https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model

2.  The application/emulation transport boundary is the emulation
component responsible for delivering messages between an emulator
instance and one or more application space processes. We call this
component a transport. Transports provide the entry and exit point for
emulator and application space messages.
https://github.com/adjacentlink/emane/wiki/Introduction#applicationemulation-transport-boundary-processing

Kaushik B. Patel
Adjacent Link LLC

Nocsaron, 2020-09-16:
Thank you for the information!  Is it possible to control EMANE's time via an event so I can run EMANE in a time-constrained environment (both much faster and significantly slower than real-time in the same scenario)

sgalgano, 2020-09-17:
It is not.

closed 2020-12-08 by Nocsaron