adjacentlink/emane #181: [Question] nems and emane daemons

bharnden, 2020-10-09:
Hello, I was hoping to get some more details in regards to the tradeoffs and pitfalls of using one EMANE daemon for multiple NEMs vs independent EMANE daemons for each NEM.

I saw this line in the FAQ, and have used multiple NEMs in the same stack before,  so I wanted to double check there are not important details being missed:
> If you are using the Virtual Transport you need to provide network stack isolation for each NEM in your EMANE deployment.

Thanks for your time!

sgalgano, 2020-10-12:
The emulator does not use lock-free data structures. Performance of any non-trivial emulation will run with better fidelity using one NEM per emulator instance and depending on the radio model, assigning specific CPUs to those emulator instances.

bharnden, 2020-10-12:
Thanks, appreciate your input.

closed 2020-10-12 by bharnden