adjacentlink/emane #183: Python 3 Installation

WorksWellWithOthers, 2020-12-18:
Is it fairly easy to setup Emane for Python 3?

sgalgano, 2020-12-18:
Yes. If you are building from source on a system with both python2 and python3 installed:

```
$ ./autogen.sh
$ ./configure --prefix=/usr PYTHON=python3
$ make install
```

If you are building emane 1.2.7 on Fedora 33, Ubuntu 18.04LTS, or Ubuntu 20.04LTS, package building is for python 3:

```
$  ./autogen.sh && ./configure && make rpm
```

or

```
$  ./autogen.sh && ./configure && make deb
```

Or you can grab the latest packages [here](https://github.com/adjacentlink/emane/wiki/Install).

WorksWellWithOthers, 2020-12-18:
Awesome. 

RHEL/CentOS 7 (x86_64)
wget https://adjacentlink.com/downloads/emane/emane-1.2.7-release-1.el7.x86_64.tar.gz

Will installing this package setup Emane for Python 3 on CentOS 7?

bharnden, 2020-12-18:
The release tarballs have both python2 and python3 packages, so yes you can set it up that way if desired.

sgalgano, 2020-12-18:
No. We only prebuild python2 support for CentOS 7. You can try rebuilding the packages from source for python3. Just modify emane.spec.in so the el7 section follows similar to el8.

WorksWellWithOthers, 2020-12-18:
Ok, sounds like I was on the right track. I find it hard to get Protobuf 3 installed correctly on my system.
I have installed most of it except for "protobuf python 3" which I need for emane python rpm.

closed 2020-12-28 by WorksWellWithOthers

WorksWellWithOthers, 2021-01-07:
[](https://github.com/adjacentlink/emane/issues/185)