adjacentlink/emane #184: EMANE CONTROL MESSAGES

rsachdeva9, 2020-12-23:
I am not clear on how to create control messages and properly read them.  I am trying to follow the IEEE wifi radio model.
 
Before calling sendDownstreamPacket(), I create a Microseconds object (Microseconds ms{10}), and then create the frequency control message:
{EMANE::Controls::FrequencyControlMessage::create(0,{{0,ms}})});
This message is created within sendDownstreamPacket().
 
After sendDownstreamPacket() is called, I try to process the packet in ProcessUpstreamPacket().  However, the controlmsg list is empty.  I am not sure what I am doing wrong or how to approach this problem.  I set the freq bandwidth to 0 because that’s how the ieee model does it.  My other question is, once the frequency control message is created, does EMANE populate or compute the rx power and other members?

sgalgano, 2020-12-23:
What do you mean by "after sendDownstreamPacket() is called, I try to process the packet in ProcessUpstreamPacket()?" Are you saying processUpstreamPacket() is being called at the receiver and you find an empty control message list?

rsachdeva9, 2020-12-23:
Thank you for the reply. Yes, processUpstreamPacket() on the receive side.

Before I call sendDownstream(), I checked wether the control message was actually created and verified it was. I even tried non zero Frequency values, to make sure if the zero value was causing a problem.  On the receive side, the control msg list is empty.

Is there some setting or parameter i need to set? I am not sure why it’s showing empty list, when the downstream data packet does successfully come through.

Ricky

sgalgano, 2020-12-23:
What does your nem xml look like?

rsachdeva9, 2020-12-23:
Our xml files are bare bone files.  The nem file contains only these parameters:

  <transport definition="transvirtual.xml">
    <param name="mask" value="255.255.255.0" />
    <param name="device" value="emane0" />
    <param name="devicepath" value="/dev/net/tun" />
    <param name="bitrate" value="1024" />
  </transport>

And the mac xml file doesn’t have any parameters.

Is there something we need to add or is required for the control messaging?

Thank you,
Ricky

sgalgano, 2020-12-23:
That is not nem xml. I suspect you are not defining your nem layers correctly.

Below is example nem xml taken from the [letce2-tutorial](https://github.com/adjacentlink/letce2-tutorial).

```
<!DOCTYPE nem SYSTEM 'file:///usr/share/emane/dtd/nem.dtd'>
<nem>
  <transport definition="transvirtual.xml"/>
  <mac definition="ieee80211abg_mac.xml"/>
  <phy>
    <param name="bandwidth" value="20M"/>
    <param name="frequency" value="2.4G"/>
    <param name="frequencyofinterest" value="2.4G"/>
    <param name="subid" value="1"/>
    <param name="systemnoisefigure" value="4.0"/>
    <param name="txpower" value="0.0"/>
    <param name="fixedantennagain" value="0.0"/>
    <param name="fixedantennagainenable" value="on"/>
    <param name="noisemode" value="outofband"/>
    <param name="noisebinsize" value="20"/>
    <param name="propagationmodel" value="precomputed"/>
  </phy>
</nem>
```

rsachdeva9, 2020-12-23:
Thanks again for the reply.  I followed that xml file (changed propagation model to freespace) and created one for our radio model.  I am still seeing empty control messages.  The EMANE version is 1.2.5.

Ricky

rsachdeva9, 2020-12-29:
Steven,  I tried using the universal phy instead of custom phy.  I am still getting empty control messages in processupstreampacket in mac layer.   I create the control message and send it downstream with data packet.

sgalgano, 2020-12-29:
With the emulator running, what does the `emanesh <emulator-address>  show` command output?

rsachdeva9, 2020-12-30:
Thanks for the reply.  I will run that command and share the output.  However, in the phy layer, when calling sendDownstreamPacket(hdr,pkt,msgs) – it seems to cause issues in EMANE.  EMANE stops working (as seen in the logs).  If I just call it without control messages – sendDownstream(hdr,pkt), It seems to work fine.  I’ve also tried the lastest emane version, still same issue.

Any idea what could be the issue there?

Thank you,
Ricky

rsachdeva9, 2021-01-04:
Steven,

Just wanted to follow up again.  When the phylayer is calling sendDownstreamPacket(hdr,pkt,msgs)

I get this error in the emane log (repeating):
15:04:39.532387 ERROR NEMOTAAdapter::processPacketQueue Excepetion caught: Not implemented
15:04:39.532861 ERROR NEMOTAAdapter::processPacketQueue Excepetion caught: Not implemented

Any idea what the error means and possible solution?

Thanks,’
Ricky

rsachdeva9, 2021-01-05:
Issue has something to do with serialization of control messages - EMANE

closed 2021-01-05 by rsachdeva9