adjacentlink/emane #185: Error while setting up LXC for Emane

WorksWellWithOthers, 2020-12-28:
```
lxc-execute: conf.c: mk_devtmpfs: 1376 /dev/.lxc is not setup - taking fallback
lxc-execute: utils.c: safe_mount: 1362 No such file or directory - Failed to mount proc onto /proc
lxc-execute: conf.c: tmp_proc_mount: 4160 No such file or directory - failed to mount /proc in the container.
scripts/demo-init: line 37: /usr/sbin/sshd: No such file or directory
Starting emane: platform1.xml
Starting emaneeventd: eventdaemon1.xml
Waiting for GPS Locatopn PTY: /emane/persist/1/var/run/gps.pty
Starting gpsd: /emane/persist/1/var/run/gps.pty
interface emane0 does not exist!
interface eth2 does not exist!
net.ipv4.conf.all.proxy_arp = 1
net.ipv4.ip_forward = 1
```

Has anyone experienced these error messages before? 
I think the emane0, eth2, is due to interfaces not existing and OLSRD is upset about it.
I'm not sure about the lxc-execute messages.

The same homebrew lxc setup script for Emane works for Python 2 but not in Python 3.

WorksWellWithOthers, 2020-12-29:
Found this in emane.log:
"ABORT FactoryException: libprotobuf.so.8: cannot open shared object file: No such file or directory"

sgalgano, 2020-12-29:
Looks like you don't have dynamic library loading configured correctly on your system or you are missing protobuf libraries. Try using `ldd` to test whether all emane dependencies are found.

```
$ ldd $(which emane)
```

WorksWellWithOthers, 2020-12-29:
It looks like my Emane is looking for so.8 but I have so.17 installed.

ldd $(which emane)
```
        linux-vdso.so.1 =>  (0x00007ffea4994000)
        libemane.so => /usr/local/lib/libemane.so (0x00007fa79f160000)
        librt.so.1 => /lib64/librt.so.1 (0x00007fa79ef58000)
        libprotobuf.so.17 => /usr/local/lib/libprotobuf.so.17 (0x00007fa79eac6000)
        libpcre.so.1 => /lib64/libpcre.so.1 (0x00007fa79e864000)
        libuuid.so.1 => /lib64/libuuid.so.1 (0x00007fa79e65f000)
        libemanexmlparser.so => /usr/local/lib/libemanexmlparser.so (0x00007fa79e44a000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fa79e22e000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007fa79e02a000)
        libxml2.so.2 => /lib64/libxml2.so.2 (0x00007fa79dcc0000)
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007fa79d9b8000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fa79d6b6000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fa79d4a0000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fa79d0d2000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fa79f596000)
        libz.so.1 => /lib64/libz.so.1 (0x00007fa79cebc000)
        liblzma.so.5 => /lib64/liblzma.so.5 (0x00007fa79cc96000)
```

Should I be building Emane to use so.17 ?

sgalgano, 2020-12-29:
You need to build emane for your target system.

WorksWellWithOthers, 2020-12-29:
```
git clone https://github.com/adjacentlink/emane.git
cd emane
git checkout tags/v1.2.6
./autogen.sh
./configure
make PYTHON=python3
sudo make PYTHON=python3 install
sudo ldconfig
```

After using the above commands, If my system has libprotobuf.so.17 would Emane be setup to use that instead of libprotobuf.so.8 ?

WorksWellWithOthers, 2020-12-31:
`git clone https://github.com/adjacentlink/emane.git
 cd emane
 git checkout tags/v1.2.6
 ./autogen.sh && ./configure
 make PYTHON=python3
 make PYTHON=python3 install
 ldconfig
 ln -s /usr/share/emane/ /usr/local/share/emane`

Is this error due to not calling, 'make rpm'?
'make: *** No rule to make target `rpm'.  Stop.'

Starting emane: platform1.xml
platform1.xml:1: I/O error : failed to load external entity "file:///usr/share/emane/dtd/platform.dtd"
<!DOCTYPE platform SYSTEM "file:///usr/share/emane/dtd/platform.dtd">
                                                                     ^
platform1.xml:2: validity error : Validation failed: no DTD found !
<platform>
         ^
01:46:19.013343 ABORT ValidateException: Failed to parse document in platform1.xml.

Possible reason(s):
 * file:///usr/share/emane/dtd/platform.dtd is missing'
 * file:///usr/share/emane/dtd/platform.dtd inaccessible (network unreachable/http down).
 * file:///usr/share/emane/dtd/platform.dtd is incorrect (run xmllint to validate).

Starting emaneeventd: eventdaemon1.xml
eventdaemon1.xml:1: I/O error : failed to load external entity "file:///usr/share/emane/dtd/eventdaemon.dtd"
<!DOCTYPE eventdaemon SYSTEM "file:///usr/share/emane/dtd/eventdaemon.dtd">
                                                                           ^
eventdaemon1.xml:2: validity error : Validation failed: no DTD found !
<eventdaemon nemid="1">
                      ^
01:46:19.533758 ABORT ValidateException: Failed to parse document in eventdaemon1.xml.

Possible reason(s):
 * file:///usr/share/emane/dtd/eventdaemon.dtd is missing'
 * file:///usr/share/emane/dtd/eventdaemon.dtd inaccessible (network unreachable/http down).
 * file:///usr/share/emane/dtd/eventdaemon.dtd is incorrect (run xmllint to validate).

b00ga, 2020-12-31:
The *rpm* make target only gets created if configure finds the `rpmbuild ` command. Is the **rpm-build** package installed?

WorksWellWithOthers, 2020-12-31:
Installing 'rpm-build' allowed me to run 'make rpm'.
After editing the 'emane.spec.in' file I'm still running into an issue related to python 2.7. I don't have anything in my 'emane.spec.in' referring to Python 2 anymore.

```
Requires: /usr/bin/python3 python(abi) = 3.6
Processing files: emane-debuginfo-1.2.6-1.el7.x86_64
Provides: emane-debuginfo = 1.2.6-1.el7 emane-debuginfo(x86-64) = 1.2.6-1.el7
Requires(rpmlib): rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1 rpmlib(Compress
edFileNames) <= 3.0.4-1
Checking for unpackaged file(s): /usr/lib/rpm/check-files /tans/emane/.rpmbuild/BUILDROOT/emane-1.2.6-1.
el7.x86_64
error: Installed (but unpackaged) file(s) found:
   /usr/lib/python2.7/site-packages/emane-1.2.6-py2.7.egg-info
   /usr/lib/python2.7/site-packages/emane/__init__.py
   /usr/lib/python2.7/site-packages/emane/__init__.pyc
   /usr/lib/python2.7/site-packages/emane/__init__.pyo
```

WorksWellWithOthers, 2020-12-31:
How can I change Python2.7 being added to the Makefile instead of Python3.6?
I tried editing 'emane.spec.in' but I'm not sure if that is what 'configure' is using to generate the Makefile. 

```
'Makefile:341:pyexecdir = ${exec_prefix}/lib64/python2.7/site-packages
Makefile:342:pythondir = ${prefix}/lib/python2.7/site-packages'
```

WorksWellWithOthers, 2020-12-31:
This is what I changed in the 'emane.spec.in'.

```
# RHEL/CentOS
%if 0%{?el7}
%define with_python3 1
```

WorksWellWithOthers, 2020-12-31:
# This directory's subdirectories are mostly independent; you can cd
# into them and run 'make' without going through this Makefile.
# To change the values of 'make' variables: instead of editing Makefiles,
# (1) if the variable is set in 'config.status', edit 'config.status'
#     (which will cause the Makefiles to be regenerated when you run 'make');
# (2) otherwise, pass the desired values on the 'make' command line.

WorksWellWithOthers, 2020-12-31:
'config.status' is generated by 'configure'.

sgalgano, 2021-01-03:
I was able to build both `python-emane` and `python3-emane` using `make rpm` after installing `python3` and `python3-devel` in CentOS 7 and adding `%define with_python3 1` to emane.spec.in:

```diff
diff --git a/emane.spec.in b/emane.spec.in
index 052f958..77e38e8 100644
--- a/emane.spec.in
+++ b/emane.spec.in
@@ -21,6 +21,7 @@ Vendor: Adjacent Link LLC
 %define with_python2 1
 %define with_old_depends 1
 %define with_pathfix 0
+%define with_python3 1
 %endif
 
 %if 0%{?el8}
```

```
$ git clone https://github.com/adjacentlink/emane
$ pushd emane
$ patch -p 1 < /path/to/above/patch/emane-add_python3_to_centos7.patch
$ ./autogen.sh && ./configure && make rpm
$ ls .rpmbuild/RPMS/noarch/python*
.rpmbuild/RPMS/noarch/python3-emane-1.2.7-1.el7.noarch.rpm
.rpmbuild/RPMS/noarch/python-emane-1.2.7-1.el7.noarch.rpm
```

You will have to make sure all the spec package requirements are correct for python3 in CentOS 7.

WorksWellWithOthers, 2021-01-05:
Thank you so much for this. I'll give this a try.

WorksWellWithOthers, 2021-01-07:
I'm having trouble using 'patch'. Is this a required step?

Would it be a problem if I directly edit the 'emane.spec.in' file to add the 'with_python3 1' line?

What I tried:
`patch -p 1 < ./new_emane.spec.in.patch`

Error: 
`patch: **** Only garbage was found in the patch input.`

WorksWellWithOthers, 2021-01-07:
'patch' seems like an important step.

```
ls .rpmbuild/RPMS/noarch/python*
.rpmbuild/RPMS/noarch/python-emane-1.2.7-1.el7.noarch.rpm
```

sgalgano, 2021-01-07:
Just manually add `%define with_python3 1` to the CentOS 7 section. Alternatively, you can create a patch file from the previously included diff output -- which does that.

WorksWellWithOthers, 2021-01-07:
I got it built this time. Not sure what happened the first time.
```
ls .rpmbuild/RPMS/noarch/python*
.rpmbuild/RPMS/noarch/python-emane-1.2.7-1.el7.noarch.rpm
.rpmbuild/RPMS/noarch/python3-emane-1.2.7-1.el7.noarch.rpm
```

WorksWellWithOthers, 2021-01-07:
```
01:33:21.866570  INFO application: emane -r -d -l 3 -f /temane/persist/1/var/log/emane.log --pidfile /emane/persist/1/var/run/emane.pid --uuidfile /emane/persist/1/var/run/emane.uuid platform1.xml
01:33:21.867284  INFO application uuid: 31b5d8e7-331a-4369-aaef-1196d23cffdf
01:33:21.874783 ABORT FactoryException: libprotobuf.so.8: cannot open shared object file: No such file or directory
```

```
ldd $(which emane)
        linux-vdso.so.1 =>  (0x00007ffd42174000)
        libemane.so => /lib64/libemane.so (0x00007fd55dad3000)
        librt.so.1 => /lib64/librt.so.1 (0x00007fd55d8cb000)
        libprotobuf.so.17 => /lib64/libprotobuf.so.17 (0x00007fd55d3e1000)
        libpcre.so.1 => /lib64/libpcre.so.1 (0x00007fd55d17f000)
        libuuid.so.1 => /lib64/libuuid.so.1 (0x00007fd55cf7a000)
        libemanexmlparser.so => /lib64/libemanexmlparser.so (0x00007fd55cd65000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fd55cb49000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007fd55c945000)
        libxml2.so.2 => /lib64/libxml2.so.2 (0x00007fd55c5db000)
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007fd55c2d3000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fd55bfd1000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fd55bdbb000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fd55b9ed000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd55df13000)
        libz.so.1 => /lib64/libz.so.1 (0x00007fd55b7d7000)
        liblzma.so.5 => /lib64/liblzma.so.5 (0x00007fd55b5b1000)
```

emane.log says it's looking for 'libprotobuf.so.8' but 'ldd' says 'libprotobuf.so.17'

I think 'libprotobuf.so.8' is for Python 2.7 so I would like emane to not be looking for that. 

Is this how it should be or am I sneaking in the wrong version of protobuf somehow?

WorksWellWithOthers, 2021-01-07:
Trying to make emane-tutorial I ran into this. 
Maybe I didn't install all the rpm packages correctly.

```
/bin/sh: line 2: emanegentransportxml: command not found
make: *** [.platform1.xml-dep] Error 127
```

WorksWellWithOthers, 2021-01-07:
```
rpm -i emane-trans*
        package emane-transportdaemon-1.2.7-1.el7.x86_64 is already installed
        package emane-trans-common-1.2.7-1.el7.x86_64 is already installed
        package emane-trans-raw-1.2.7-1.el7.x86_64 is already installed
        package emane-trans-virtual-1.2.7-1.el7.x86_64 is already installed
```

WorksWellWithOthers, 2021-01-07:
It seems I'm not installing Protobuf correctly.

```
wget https://rpmfind.net/linux/opensuse/distribution/leap/15.1/repo/oss/x86_64/python3-protobuf-3.5.0-lp151.4.7.x86_64.rpm
yum localinstall python3-protobuf*.rpm
```
```
Examining python3-protobuf-3.5.0-lp151.4.7.x86_64.rpm: python3-protobuf-3.5.0-lp151.4.7.x86_64
python3-protobuf-3.5.0-lp151.4.7.x86_64.rpm: does not update installed package.
Nothing to do
```

```
checking for protobuf... no
configure: error: Package requirements (protobuf) were not met:

No package 'protobuf' found
```

WorksWellWithOthers, 2021-01-07:
```
checking for protobuf... no
configure: error: Package requirements (protobuf) were not met:

No package 'protobuf' found
```

WorksWellWithOthers, 2021-01-07:
I think this resolved the [ ./configure](url) issue.

```
ls /usr/lib
libprotobuf.a
libprotobuf.so
libprotobuf.so.17

echo $PKG_CONFIG_PATH
/usr/lib

export PKG_CONFIG_PATH=/usr/lib/pkgconfig
```

WorksWellWithOthers, 2021-01-08:
Do I not have python3-protobuf installed?

```
yum localinstall python3-emane*
Loaded plugins: fastestmirror, ovl
Examining python3-emane-1.2.7-1.el7.noarch.rpm: python3-emane-1.2.7-1.el7.noarch
Marking python3-emane-1.2.7-1.el7.noarch.rpm to be installed
Resolving Dependencies
--> Running transaction check
---> Package python3-emane.noarch 0:1.2.7-1.el7 will be installed
--> Processing Dependency: python3-protobuf for package: python3-emane-1.2.7-1.el7.noarch
Loading mirror speeds from cached hostfile
 * base: la.mirrors.clouvider.net
 * extras: mirrors.raystedman.org
 * updates: mirror.san.fastserv.com
--> Finished Dependency Resolution
Error: Package: python3-emane-1.2.7-1.el7.noarch (/python3-emane-1.2.7-1.el7.noarch)
           Requires: python3-protobuf
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest
```

WorksWellWithOthers, 2021-01-08:
That seems off. 

```
pip3 list
protobuf               3.14.0
python3-protobuf       2.5.0
```

WorksWellWithOthers, 2021-01-08:
```
protobuf.x86_64                         3.6.1-4.el7                    @rpmrepo
protobuf-compiler.x86_64                3.6.1-4.el7                    @rpmrepo
protobuf-devel.x86_64                   3.6.1-4.el7                    @rpmrepo
```

WorksWellWithOthers, 2021-01-08:
```
Installed /usr/local/lib/python3.6/site-packages/protobuf-3.14.0-py3.6.egg
Processing dependencies for protobuf==3.14.0
Searching for six==1.15.0
Best match: six 1.15.0
Adding six 1.15.0 to easy-install.pth file

Using /usr/local/lib/python3.6/site-packages
Finished processing dependencies for protobuf==3.14.0
```

b00ga, 2021-01-08:
Part of the problem with trying to use the combination of EMANE / CentOS 7 / python3 is that the emane python bindings want the python bindings for protobuf. But if you're trying to do it all with RPMs, there's no python3-protobuf for EL7 (there are for EL8 and Fedora).

I experimentally played around with this and used some tooling for RPM generation to create a python3-protobuf RPM matching the protobuf version that ships with CentOS7 (which getting that tooling has it's own dependency issues).

Anyway, if you are interested in trying it, I wrote up my instructions in this gist:
https://gist.github.com/b00ga/9ffcafba7562d35ebeccf4659df783bf

WorksWellWithOthers, 2021-01-08:
Thank you. I'll give this a try.

WorksWellWithOthers, 2021-01-08:
I was able to install emane for Python3  following the instructions.
 These were the dependencies I had to install.

```
gcc
make
wget

wget https://rpmfind.net/linux/centos/8-stream/AppStream/x86_64/os/Packages/python3-lxml-4.2.3-1.el8.x86_64.rpm
yum localinstall python3-lxml*.rpm
```

WorksWellWithOthers, 2021-01-08:
```
python3 -c 'import emane.info; print(emane.info.__version__)'
1.2.7
```

```
>>> import emane
>>> import emane.shell
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/emane/shell/__init__.py", line 34, in <module>
    from .controlportclient import ControlPortClient
  File "/usr/lib/python3.6/site-packages/emane/shell/controlportclient.py", line 34, in <module>
    from . import remotecontrolportapi_pb2
  File "/usr/lib/python3.6/site-packages/emane/shell/remotecontrolportapi_pb2.py", line 317, in <module>
    has_default_value=False, default_value=unicode("", "utf-8"),
NameError: name 'unicode' is not defined
```

WorksWellWithOthers, 2021-01-08:
```
Python 2.7.5 (default, Nov 16 2020, 22:23:17)
[GCC 4.8.5 20150623 (Red Hat 4.8.5-44)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import emane
>>> import emane.shell
>>>
```

WorksWellWithOthers, 2021-01-08:
Emane-Tutorial seems to be good.

```
./demo-start
Creating bridge: emanenode0
waiting for emanenode0...found
kernel.sched_rt_runtime_us = -1
Starting lxc instance: node-1
Starting lxc instance: node-2
```

WorksWellWithOthers, 2021-01-08:
Am I doing something wrong with the ControlPoint Client / emane.shell ?

b00ga, 2021-01-09:
re: lxml. I basically don't run any RHEL/CentOS systems without add the EPEL repository (https://fedoraproject.org/wiki/EPEL#Quickstart). Python3's lxml will get automatically pulled in as a dependency if EPEL is installed.

Re: python3 emane.shell stackrace and unicode error.
This stackoverflow post (https://stackoverflow.com/questions/38697037/how-to-convert-python-2-unicode-function-into-correct-python-3-x-syntax) seems to indicate that the unicode() function is a python2-ism. The file with the error (emane/shell/remotecontrolportapi_pb2.py) is python code generated from protobuf bindings. If I had to speculate, protobuf 2.5.0 didn't have fully fleshed out python3 support yet.

Building newer protobuf is maybe an option. I did a quick attempt rebuild the CentOS8 protobuf source rpm but it failed.

Is python3 really required in your case? Everything in C7 should work fine with the python2 support. Or the needed packages should be available on CentOS8 if that's an option instead (there's a python3 protobuf package in the distro repos).

b00ga, 2021-01-09:
@WorksWellWithOthers Be careful with random RPM packages from RPMfind.

https://rpmfind.net/linux/opensuse/distribution/leap/15.1/repo/oss/x86_64/python3-protobuf-3.5.0-lp151.4.7.x86_64.rpm
is a package for OpenSUSE, a completely different linux distro from CentOS and probably why iot did not work.

https://rpmfind.net/linux/centos/8-stream/AppStream/x86_64/os/Packages/python3-lxml-4.2.3-1.el8.x86_64.rpm
is a package for CentOS Stream 8. Typically you do not want to install packages from. a different major version of RHEL/CentOS (EL7 vs EL8 in this case). Usually the major versions can have different and incompatible versions of things like the main C library (libc), different versions of yum & rpm that can be incompatible, and can require newer dependencies that exist in the new major version than exist in the older major version.

WorksWellWithOthers, 2021-01-12:
I rebuilt all the dependencies using 'python3-lxml' from the EPEL repository.
Ended up running into the same issue so I think you are correct about the 'protobuf 2.5.0' being the suspect. 
I'll attempt to build a newer protobuf ( I think I tried this already before ). It might have been a 'emane shell' works but 'emane' didn't because I did other stuff wrong.  

`Is python3 really required in your case?`
It's not required. Python 2 works really well.

I was thinking of using a CentOS8 image but got scared of their recent EOL announcement.

8 EOL 2021.
7 EOL 2024. 

Maybe I should transition to Stream 8?

WorksWellWithOthers, 2021-01-12:
CentOS 8 Stream
```
checking for protobuf... no
configure: error: Package requirements (protobuf) were not met:

Package 'protobuf', required by 'virtual:world', not found
```

`/usr/lib64/libprotobuf.so.15`

`export PKG_CONFIG_PATH=/usr/lib64/pkgconfig`

export doesn't seem to help ./configure find the .so

CentOS 7
I thought I could just use the same instruction but change the version to the latest.

```
fpm -s python -t rpm --version 3.15.0 --python-bin python3 --python-easyinstall e
asy_install-3.6 --python-package-name-prefix python3 --no-python-fix-name python3-protobuf
Doing `require 'backports'` is deprecated and will not load any backport in the next major release.
Require just the needed backports instead, or 'backports/latest'.
Process failed: easy_install-3.6 failed (exit code 1).
```

b00ga, 2021-01-13:
* python3-protobuf not being newer than 2.5.0
So I noticed on the pypi page for this package that it only has a 2.5.0 release (https://pypi.org/project/python3-protobuf/#history). I suspect someone created this expressly for EL7 support.
* Py2/Py3 in EL7
RHEL7.0 was released in 2014 and only came with Python2. Python3 was later bolted on with the Software Collections subsystem, and eventually a non-SCL based Python3 was added in RHEL7.7 (August 2019). So you can see that Py3 didn't actually drop until 5 years after RHEL7 released, and the Py3 support isn't 100% on par with the Py2 support. EL8 comes with Py3 out-of-box.
* CentOS 8 vs CentOS Stream 8
CentOS 8 is a rebuild of RHEL 8 from the the RHEL 8 sources, which is referred to as a _downstream_ distribution. CentOS Stream 8 is basically the next release of RHEL before it comes out, basically current RHEL version plus packages that have passed Red Hat QA. The goal is that the community can now contribute to what comes out in the next version of RHEL. In this capacity, CentOS Stream acts as an _upstream_ to RHEL. The average user won't noticed a difference between CentOS 8 and CentOS Stream 8.
* Missing protobuf in CentOS (Stream) 8
Ok, Red Hat used to have a developer extras repo called Power Tools in EL7 and earlier. In EL8, Red. Hat renamed this CodeReady Builder and moved some of the _*-devel_ RPMs that used to come with the base system into this new repo. CentOS retained the older PowerTools name, but they don't enable it by default. You need to enable it to get some of those devel packages that are needed for emane.

Building RPMs on EL8 should be as (relatively) simple as:
```
export EMANE_VER=1.2.7

wget --content-disposition https://github.com/adjacentlink/emane/archive/v${EMANE_VER}.tar.gz

tar xvfz emane-${EMANE_VER}.tar.gz

cd emane-${EMANE_VER}

sudo yum-config-manager --enable PowerTools

sudo dnf install -y wget gcc gcc-c++ make autoconf automake libtool libxml2-devel protobuf-devel pcre-devel libuuid-devel libpcap-devel rpm-build python3-devel

./autogen.sh
./configure

make rpm
```

WorksWellWithOthers, 2021-01-13:
Thank you for taking the time to write a detailed response. I am learning a lot.

WorksWellWithOthers, 2021-01-13:
```
yum update
yum install yum-utils
yum-config-manager --enable powertools
dnf install -y wget gcc gcc-c++ make autoconf automake libtool libxml2-devel protobuf-devel pcre-devel libuuid-devel libpcap-devel rpm-build python3-devel
export EMANE_VER=1.2.7
wget --content-disposition https://github.com/adjacentlink/emane/archive/v${EMANE_VER}.tar.gz
tar xvfz emane-${EMANE_VER}.tar.gz
cd emane-${EMANE_VER}
./autogen.sh
./configure
make rpm
```

```
Python 3.6.8 (default, Aug 24 2020, 17:57:11)
[GCC 8.3.1 20191121 (Red Hat 8.3.1-5)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import emane
>>> import emane.shell
>>>
```

closed 2021-01-13 by WorksWellWithOthers