adjacentlink/emane #187: python3-pynodestatviz install issue from .deb on Ubuntu 20.04 - No python3-pmw available

bullr, 2021-01-19:
Installing EMANE from .deb files on a clean install of Ubuntu 20.04 server.

Following the wiki:
dpkg -i *.deb

Installs everything fine after I worked out the missing dependencies except for: python3-pynodestatviz

dpkg -i python3-pynodestatviz_0.2.3-1_all.deb 
Selecting previously unselected package python3-pynodestatviz.
(Reading database ... 103450 files and directories currently installed.)
Preparing to unpack python3-pynodestatviz_0.2.3-1_all.deb ...
Unpacking python3-pynodestatviz (0.2.3-1) ...
dpkg: dependency problems prevent configuration of python3-pynodestatviz:
 python3-pynodestatviz depends on python3-pmw; however:
  Package python3-pmw is not installed.

dpkg: error processing package python3-pynodestatviz (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 python3-pynodestatviz

Things I have tried:
- apt-cache search python3-pwm returns empty
- however there is a python-pwm package, installing does not help since it seems to be python2 specific
- Installing python3-pip, then running pip3 install pmw installs Pmw 2.0.1 which still does not get recognized by the package installer.

pip3 install pmw
Requirement already satisfied: pmw in /usr/local/lib/python3.8/dist-packages (2.0.1)
(just tossing for reference after I installed it)

bullr, 2021-01-20:
Building pynodestatviz from the source code available on github did the trick.  It used the python-pmw installed by apt on Ubuntu 20.  The .deb pre-packaged with the emane release needs to be fixed to recognize the correct dependency name it seems (ie. python3-pmw -> python-pmw).

geraldolsribeiro, 2022-09-06:
A workaround until the dependency changes:

```bash
#!/bin/bash
# Repack the Debian package
DEB=python3-pynodestatviz_0.2.4-1_all.deb
if [ ! -f "$DEB.bak" ];
then
  cp "$DEB" "$DEB.bak"
  ar x "$DEB"
  tar xvf control.tar.xz
  sed -i "s/python3-pmw/python-pmw/" control
  tar cJf control.tar.xz postinst prerm md5sums control
  ar rcs $DEB debian-binary control.tar.xz data.tar.xz  # the order is important here
  rm data.tar.xz control.tar.xz postinst prerm md5sums control 
fi
```

Lithimlin, 2022-12-12:
> A workaround until the dependency changes

Where exactly does this need to be executed? From within the `.debbuild` directory?