adjacentlink/emane #188: ABORT unable to set realtime scheduler

lemon-lyman, 2021-02-19:
I'm unable to set realtime scheduler. When running the command:

`emane -d -l 4 -r -f /tmp/pycore.1/CoreNode2.conf/CoreNode2-emane.log /tmp/pycore.1/CoreNode2.conf/CoreNode2-platform.xml `

The output in the log looks like:

```
root@CoreNode2:/tmp/pycore.1/CoreNode2.conf# cat CoreNode2-emane.log 
13:39:52.544491  INFO application: emane -d -l 4 -r -f /tmp/pycore.1/CoreNode2.conf/CoreNode2-emane.log /tmp/pycore.1/CoreNode2.conf/CoreNode2-platform.xml 
13:39:52.544618  INFO application uuid: a65da7fa-4489-468b-a878-de86688c8ed6
13:39:52.544657 ABORT unable to set realtime scheduler
```

Is it possible to print a more descriptive error message than this? Unsure how to debug this with only that abort message.

I'm using EMANE 1.2.5 and CORE 7.2.1 both inside a ubuntu docker container. The CoreNode2-platform.xml looks like:

```
<?xml version='1.0' encoding='UTF-8'?>
<!DOCTYPE platform SYSTEM "file:///usr/share/emane/dtd/platform.dtd">
<platform>
  <param name="antennaprofilemanifesturi" value=""/>
  <param name="controlportendpoint" value="0.0.0.0:47000"/>
  <param name="eventservicedevice" value="ctrl0"/>
  <param name="eventservicegroup" value="224.1.2.8:45703"/>
  <param name="eventservicettl" value="2"/>
  <param name="otamanagerchannelenable" value="1"/>
  <param name="otamanagerdevice" value="ctrl0"/>
  <param name="otamanagergroup" value="224.1.2.8:45702"/>
  <param name="otamanagerloopback" value="0"/>
  <param name="otamanagermtu" value="0"/>
  <param name="otamanagerpartcheckthreshold" value="2"/>
  <param name="otamanagerparttimeoutthreshold" value="5"/>
  <param name="otamanagerttl" value="1"/>
  <param name="stats.event.maxeventcountrows" value="0"/>
  <param name="stats.ota.maxeventcountrows" value="0"/>
  <param name="stats.ota.maxpacketcountrows" value="0"/>
  <nem id="1" name="tap2.0.1" definition="eth0-nem.xml">
    <transport definition="eth0-trans-virtual.xml">
      <param name="device" value="eth0"/>
    </transport>
  </nem>
</platform>
```

sgalgano, 2021-02-19:
The user running emane does not have privileges to set the realtime scheduler, which is required because of the specified `-r`.

lemon-lyman, 2021-02-19:
This ended up being a docker issue more than anything. My container already had full privileges and capabilities but I needed to enable realtime scheduling within the container: [https://stackoverflow.com/questions/46563332/docker-daemon-container-real-time-scheduling-with-ubuntu-linux-host]

sysctl -w kernel.sched_rt_runtime_us=-1

closed 2021-02-19 by lemon-lyman