adjacentlink/emane #189: Physical layer frequency and frequencyofinterest setting in FDMA multi-hop network

ChengDaHaI, 2021-02-22:
Hi,

I want to use RFpipe to emulate a multi-hop network where each hop use one transmitting frequency.

For example, a four nodes case: "SRC --> RLY1 --> RLY2 --> DST". 
I set: 

- frequency of SRC to be 2.4GHz and frequencyofinterest to be 2.3GHz 
- frequency of RLY1 to be 2.5GHz and frequencyofinterest to be 2.4GHz
- frequency of RLY2 to be 2.6GHz and frequencyofinterest to be 2.5GHz

However, i find something strange:

- RLY1 **CANNOT** communicate with RLY2 by using python UDP socket (sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)).
- RLY1 **CAN** communicate with RLY2 by using python UDP socket in broadcast mode (sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)).
- If I set frequencyofinterest of RLY1 to be 2.6GHz , RLY1 can communicate with RLY2 using python UDP socket without broadcast mode.

Any suggestion on how to emulate a multihop FDMA wireless network?

Thanks!

ChengDaHaI, 2021-03-05:
Any comments?

sgalgano, 2021-03-05:
My guess is RLY1 is not receiving RLY2's arp response. You can try manually populating the arp cache on RLY1, so it does not arp for RLY2.